arkhipenko/TaskScheduler #101: Unable to pass class member funtion into Task

nxg3n, 2020-07-16:
I am writing a class that will manage the tasking side of my program. However, when I tray to instantiate a new Task object by passing in &SomeClass::startDayTask I get the following error.

Task startDayTask(1000, 10, &SomeClass::startDay);

Same error for:

Task startDayTask(1000, 10, &startDay);

error: no matching function for call to 'Task::Task(int, int, void (SomeClass::*)())'
     Task startDayTask(1000, 10, &SomeClass::startDay);

nxg3n, 2020-07-16:
Also to clarify this code resides inside a SomeClass method.

arkhipenko, 2020-07-16:
You can not use dynamically created objects as a reference with a static task.  How would a compiler know which instance of &SomeClass::startDay to pass to? 

If you want to use dynamic binding you have to compile with `#define _TASK_OO_CALLBACKS` compile option. 
There is not much documentation for it yet - did not have enough time to create one. 

It is completely different approach where instead of declaring tasks as static objects you create your specific tasks as objects which you inherit from the class "Task". 

This example should give you an idea:
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example21_OO_Callbacks

I personally find this approach more trouble than benefit, but there is at least one guy who likes it a lot.

As an ugly workaround, you can pass the object pointer via LTS to a task, but that is, as I said, an ugly workaround. 
https://github.com/arkhipenko/TaskScheduler/wiki/API-Task#local-task-storage-methods-1
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example08_LTS

closed 2020-07-16 by arkhipenko

nxg3n, 2020-07-16:
Hi @arkhipenko ,

Thanks so much for the response. I will take a look at the examples now.

I am doing it this way as I have a control system in which i have various different schedule possibilities. Each of these and its params are stored in an object. This allows me to init these objects at compile-time and cycle through the options on the keypad.

I Will let you know how I go.

GitMoDu, 2020-07-17:
> but there is at least one guy who likes it a lot.

Hi!