arkhipenko/TaskScheduler #102: Overwriting Arduino's "delay" function?

FrankyBoy, 2020-08-04:
Hi there!

Is it possible to overwrite arduino's built-in "delay" function with the non-blocking one supplied by this library? I.e. basically doing the same as Arduino's own Scheduler does? Would be really helpful together with other libraries that just use delay() internally. 

Cheers :)

arkhipenko, 2020-08-04:
No. It's a completely different context. TS's delay is a delay of a single Task, while Arduino delay is a delay of the entire sketch.

closed 2020-08-04 by arkhipenko

FrankyBoy, 2020-08-04:
As far as I can tell, delay(), if you use it in the context of a Scheduler.h thread, is supposed to yield the thread context.

FrankyBoy, 2020-08-04:
Alright, did some digging,  hope I found the right code ... 

* Arduino code [yields in delay](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/delay.c#L64), just so we have a baseline reference
* ESP32 [calls vTaskDelay](https://github.com/espressif/arduino-esp32/blob/c2b3f2d6afc1911db1974a324782730dd642d524/cores/esp32/esp32-hal-misc.c#L144) which [I have no damn idea what this does](https://github.com/aws/amazon-freertos/blob/f33863740032c59307b1849629c736a416ea2687/vendors/nordic/nRF5_SDK_15.2.0/external/freertos/source/tasks.c#L1288), but it sounds like it would also yield. 
* ESP8266 ... tbh I'm not gonna dig now, it's after midnight here. 

Would that mean delay() actually already _should_ yield to another thread (which would be nice)?

arkhipenko, 2020-08-04:
TaskScheduler is a **cooperative** library. All tasks run in the same thread. There is no yield to "other" thread. You "yield" or rather just return from the callback method back to the scheduler to pass control to another task. That's the principle difference.

FrankyBoy, 2020-08-05:
Arduino's scheduler is also cooperative. Otherwise you wouldn't have to yield() in the first place. I is quite literally exactly the same concept.

arkhipenko, 2020-08-05:
> It is quite literally exactly the same concept.

I respectfully disagree. Although in principle Scheduler does implement cooperative scheduling, the fundamental difference is that TS does not do context switching. TS actual scheduler part is small and efficient precisely because it does not maintain the task's state (registers and stack). 
Also the yielding part (do have a look under the hood of Scheduler!) is implemented in assembly and is very platform-dependent, while TS runs practically on anything (even Raspberry Pi - I did that!) 

I see the allure of delay() - it does not require breaking up code into chunks. Unfortunately on the scheduling side, it creates too many headaches. 

So here, unfortunately, I will not be overloading the delay(), and anyone is free to use Scheduler (if you are running on SAM or SAMD board).   In TS you yield when you return from the task's callback.

By the way - a positive side effect of running in the same thread - you only need 1 stack, and you need to worry about 1 stack. With Scheduler (and FreeRTOS for that matter) you have to provide stack memory for every task, which is a) waste, and b) additional headache since now you need to worry about stack overruns as many times as the number of tasks you run. 

And finally, the API of Scheduler is a joke. 3 methods?... 

Cheers