arkhipenko/TaskScheduler #106: [ESP32] getCpuLoadIdle() always return 0

Schrolli91, 2020-10-07:
I testing this lib on a TTGO T-Beam V1.1 with an ESP32 chip.
I want to calculate the cpu load that i can estimate, how much resources i have over atm.

But i cant get the idle ticks because `getCpuLoadIdle()` always return 0.

Do anyone have a idea, what i can check?

```
#define _TASK_TIMECRITICAL       // Enable monitoring scheduling overruns
#define _TASK_PRIORITY           // Support for layered scheduling priority
#define _TASK_OO_CALLBACKS       // Support for dynamic callback method binding
```

main.cpp includes `TaskScheduler.h` all other files the `TaskSchedulerDeclarations.h`.
The config defines are located before each of this includes.

Thanks and regars,
Basti

arkhipenko, 2020-10-07:
I assume 
#define _TASK_SLEEP_ON_IDLE_RUN  defined as well. That is a pre-requisite. 

For ESP32 you also need to implement some kind of sleep mode for the MCU to actually go into idle state. 
As of v 3.2.0 all idle sleep methods are moved to a premium feature set (meaning I would gladly implement them for you for a small fee). 
Alternatively, you can implement a method like this one below (this is for AVR boards):

```
#if defined( ARDUINO_ARCH_AVR ) // Could be used only for AVR-based boards.

#include <avr/sleep.h>
#include <avr/power.h>

void SleepMethod( unsigned long aDuration ) {
  set_sleep_mode(SLEEP_MODE_IDLE);
  sleep_enable();
  /* Now enter sleep mode. */
  sleep_mode();

  /* The program will continue from here after the timer timeout ~1 ms */
  sleep_disable(); /* First thing to do is disable sleep. */
}
// ARDUINO_ARCH_AVR
```
and then assign it to the scheduler via:
`void setSleepMethod( SleepCallback aCallback );`

Note on esp32 - I found implementing light sleep working in many cases, but sometimes it affected the WiFi connection. I did not have time to experiment further. 

Good luck!

Schrolli91, 2020-10-07:
> Alternatively, you can implement a method ...
> and then assign it to the scheduler via:
> `void setSleepMethod( SleepCallback aCallback );`

Thanks for this hint - works for me now.
Maybe that should be documented (ESP behavior) - or did I just not see it?

closed 2020-10-07 by Schrolli91