arkhipenko/TaskScheduler #107: Problem with TaskScheduler and HardwareSerial

Schrolli91, 2020-10-09:
I want to get GPS data from an NEO6N GPS module.
That alone works fine and i get the GPS NEMA strings from the second serial interface on my ESP32.

But when im using the TaskScheduler and call `sched.execute();` perdiodicly i only got these cryptic output:

```
$GPGSV,۲s�v�W␂��b␌      $%�R��R3D␚i*2[�� ␂␂���
c␂��K�&&b���bR�␒5
$GP��
0�vT0a␂␆���h!␂2��ē␙b�j�B3"␂�L4L0#��si␂�O��b���b��b��b* ␃b`��2�a␀��␐)"��b�)�dh&3��"2�ٚ�"␀k␂S!(&20,29,04,58,06!(
���NWH"���␄a"��ɉHb␒��@@b&&��kiH ����HB␅4�,110,30,04,58��ѶB␀b�b]6␐ ␂␆�9�iەII��   �`�4�s/Hd�R-*␕ib�b��b��b��␅��r␐�␀␂�R0␒␇X&�ёK!�͂!␂��:H b␅گ␐!b�␑�2$(b'␓b���b��b��b��R␒@k␀���� ␂2��6␃b�␂��'␄A#��MH��R� ␄!f��i�␀"�͓␓Ɗ�b��b��b���۩2"␋ �"�&␋��2��+␒�̆�*�}IM J;␡j␅mb␒s+D␆␀b���␔uMY�3,1,12,�@j"�␃�8 "��dP�b���␕K#�ͨ`␔i␂F�2␃�i␂�␋�> "��GvH�:A␝MY�3,1,12��A␀j��S�
j␐�␓)r��R��j
$GP�H�I␒r-B�)␂&rI2J)"�-␛��S�� B�}�  
```

So you can see, some chars are ok, but most of them are corrupted. Looks like TaskScheduler will interference with the serial comm.
Do anyone have any ideas whats going wrong here? Or is this a TaskScheduler restriction?

arkhipenko, 2020-10-09:
What compile parameters are you using? What version of TS are you on? 
Can you send me the code?

Schrolli91, 2020-10-09:
I had found the problem now... For others with the same Problem:

The ESP32 light sleep in my own set `void sleepCallback (unsigned long aDeltaTime)` destroyed the timing *facepalm*

@arkhipenko Thank you for your quick response anyway

closed 2020-10-09 by Schrolli91

arkhipenko, 2020-10-09:
I suspected that but wanted to make sure, hence the questions. Good luck!