arkhipenko/TaskScheduler #108: Multiple definition issue when using multiple header/.h files

michaelfransen, 2020-10-10:
I cannot compile my project if I import `TaskScheduler.h` on multiple C++ header files. If I just import it on one I am fine. This is the error I get. 

```
.pio/build/esp32dev/src/microapps/WeatherApp.cpp.o: In function `Scheduler::allowSleep(bool)':
WeatherApp.cpp:(.text._ZN9Scheduler10allowSleepEb+0x0): multiple definition of `Scheduler::allowSleep(bool)'
```

I see this issue was once fixed with https://github.com/arkhipenko/TaskScheduler/commit/59bff57af2d3c326ed92c315d82fdbbeebfbaf17 but now it appears all those inlines are gone. How can one use TaskScheduler with multiple header files?

closed 2020-10-10 by michaelfransen

thijstriemstra, 2021-01-12:
I'm seeing the same errors, for example:

```
multiple definition of `Scheduler::execute()'
```

and unable to compile the OOP example. How did you manage to solve this @michaelfransen?

arkhipenko, 2021-01-12:
Have a look at issue #93