arkhipenko/TaskScheduler #109: Time when next Task should be run

Schrolli91, 2020-10-10:
I think thats not possible at the moment - but maybe a nice-to-have feature.

Is it possible to get the time from scheduler, when the next task should be run?
If i know, when next task should be run, i can sleep the until this time.
Maybe this can reduce power consumption a lot.

I have tested this in a little proof of concept where i calc the shortest time to the next time in the sleep callback.
At a first look it looks like working properly - maybe thats a nice feature for the future?

Have you thought about such a function till now - and do you think that there are some architecture problems for code this in a fast and stable way?

arkhipenko, 2020-10-10:
I did think about this. It would work only in very simple cases, and there are other issues with putting MCU to sleep for longer than 1 ms: 

- TS relies on millis() to keep track of time. Most MCUs that do not have RTCs will lose track of time if put to sleep for longer since millis() will not be updated and when MCU wakes up it will think no time passed, so the scheduling will be off. YOu can argue that you know the sleep interval and can update the tasks schedule accordingly. Yes, **BUT** if MCU is woken up by an interrupt before the sleep interval was over, you cannot be sure that exactly what time passed and the scheduling will be off. 
- TS has a concept of event-driven task invocation (StatusRequest objects). There is no actual "start time" for those as they are just waiting for an event. If you put MCU to sleep the event maybe missed. 

Since I have to maintain TS for general purpose I chose not to implement longer sleep intervals. 

You can use this method:
https://github.com/arkhipenko/TaskScheduler/wiki/API-Task-Scheduler#long-timeuntilnextiteration-task-atask
for simple cases with just a few tasks, but remember - you need to reset the scheduling after wake up since millis() will be off.

closed 2020-10-10 by arkhipenko

Schrolli91, 2020-10-10:
thank you for your comments. I think you are right with the points given.