arkhipenko/TaskScheduler #113: correct setting of tasks

DavidAntonin, 2021-01-16:
Hello, what is the best way to solve this with taskscheduler?
When temp is below 0°C  run water pump every 90 minutes for 10 minutes.
My idea is have a 2 tasks
Task 1 enebled if  temp is 0, interval 90*60*1000 run forever - start pump
Task 2 enabled with taks 1, interval 10*60*1000 run only once
I have a problem with task 2 - how to correctly start it from task 1?
Thank you

TD-er, 2021-01-16:
Why use 2 periodical tasks for this?
What you want to do is to make sure the pump doesn't start/stop too often, so when it starts it should run for 10 minutes.
If it turns off, you should schedule a new task to run in 80 minutes.

Task 1: (pump is off)
- If temp > 0: schedule Task 1 in 1 minute from now (not periodical)
- Else: Turn on pump and schedule task 2 to run in 10 minutes (to turn off the pump, not periodical)

Task 2: (pump is on)
- turn off pump
- Schedule task 1 (not periodical) in 80 minutes.

DavidAntonin, 2021-01-16:
I do not have to periodical tasks. Task2 for pump off is TASK_ONCE and is enableDelayed from Task1 which switch pump on

arkhipenko, 2021-01-16:
I would use one task and getRunCounter method. On odd counters start the pump and set interval for 10 minutes. On even counters, set interval for 80 minutes and turn pump off. 

You should really look at the blink0 example. It gives you many scheduling ideas.

arkhipenko, 2021-01-16:
Odd/even is just bit0, so it's 
Task.getRunCounter()&1

TD-er, 2021-01-16:
Using the last bit of the set counter is a nice one!

However, you must keep in mind the period after switching off the pump should be 80 minutes, but every other check after it where the temp is > 0 C should also set an odd counter but for a shorter interval (e.g. 1 minute) or else you may get in a cycle where the temperature gets below 0 but the pump is not turned on.

closed 2021-01-18 by arkhipenko