arkhipenko/TaskScheduler #115: How to allocated scheduler as static member

jaccobezemer, 2021-02-01:
Hello,

I'm trying to create a class that uses a static declared scheduler but i'm getting undefined reference to `Scheduler::Scheduler()
The other classes that are declared static don't give an error.

in pump.h
```
class Pump {
public:
    static uint8_t bus_;
    static OneWire oneWire_;     
    static DallasTemperature sensors_;
    static DeviceAddress flowThermometer;
    static Scheduler ts;
    static void start();
    static void loop();
}
```
in pump.cpp
```
OneWire Pump::oneWire_ = OneWire(ONE_WIRE_BUS);
DallasTemperature Pump::sensors_ = DallasTemperature(&oneWire_);
DeviceAddress Pump::flowThermometer;
Scheduler Pump::ts;
```

How can I allocate a statically declared Scheduler class? 

Kind regards,
Jacco

TD-er, 2021-02-01:
Why declare these functions and member variables `static`?

A static function is one that doesn't need to access the members.
For example a function that only does conversions and has all it needs given via parameters.

A static member is one that's shared on all instances of that class.
Is that what you try to achieve here?
Why you even declare it in the .h file? It is probably working fine if you only declare it in the .cpp file, which also gives you some safety as others can't directly access the object if they have an instance (or pointer to) the `Pump` class.

arkhipenko, 2021-02-02:
I second @TD-er concern. Why static member methods? 

All I can recommend is to read the Wiki
https://github.com/arkhipenko/TaskScheduler/wiki/Creating-TaskScheduler---friendly-libraries

PainlessMesh is a great example of a library that uses TaskScheduler. I just finished a project using it and it was awesome to be able to coexist with a library that uses cooperative multi-tasking the right way.

closed 2021-02-02 by arkhipenko

jaccobezemer, 2021-02-02:
I'm using example code form another framework/library.
This example uses static members. Without changing to much (i'm just a hobbyist and don't know ) I wan't to embed TaskScheduler.
So, there is no way using TaskScheduler as a static member?
Thanks anyway.

TD-er, 2021-02-02:
You can use it as a static member, but I really wonder why as it is probably not what you want as static members serve a very specific use case.