arkhipenko/TaskScheduler #118: TaskScheduler in custom library

R1Snake, 2021-03-18:
Hi,

i want to declare and define a TaskScheduler in my custom library but I have a problem with the declaration.

This ist my Code:

TaskExample.h:

`#ifndef TaskExample_h
#define TaskExample_h

#include <TaskScheduler.h>

class TaskExample
{
    public:
    TaskExample();
    void defineMyTask();
    void taskToExecute();

    Task *myTask;
    Scheduler ts;

};

#endif`

TaskExample.cpp:

`#include "TaskExample.h"

TaskExample::TaskExample(){}

void TaskExample::defineMyTask(){
    myTask = new Task (500,TASK_FOREVER, TaskExample::taskToExecute, &ts, false); // Funktioniert nicht (Falsche Parameter)
    myTask = new Task (500,TASK_FOREVER, externalVoid, &ts, false); // Funktioniert, bringt mir jedoch nichts
}

void externalVoid(){
    Serial.println("Ich werde ausgeführt, habe aber keinen Klassenbezug");
}

void TaskExample::taskToExecute(){
    Serial.println("Ich werde ausgeführt");
}

`

In "defineMyTask" i got an error because i have the wrong parameters and the second line works fine. What I have to do to enter my library void into the declaration?

arkhipenko, 2021-03-19:
I am not sure what you are trying to achieve, therefore it is hard to help.

Generally, TaskScheduler is used with static Task definitions in a non-class implementation. 
There is a way to implement Tasks in a true OO style: there is no documentation, but you have to study and follow example #21:
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example21_OO_Callbacks

If you want to include TaskScheduler in your library, the best example is pailessMesh:
https://gitlab.com/painlessMesh/painlessMesh
it uses advice from
https://github.com/arkhipenko/TaskScheduler/wiki/Creating-TaskScheduler---friendly-libraries

Hope this helps.
Regards
Anatoli

R1Snake, 2021-03-19:
My target is only to execute a library void with the task scheduler. 

It is important that the TaskScheduler is defined and declared in the library because I want the library "standalone" and the task has to work with the library objects.

arkhipenko, 2021-03-19:
I am sorry, I do not understand what "execute a library void" means.

R1Snake, 2021-03-19:
Work with xTaskCreate because i get an error that Task is already defined if I include your library in my custom library and in main.cpp.

closed 2021-03-19 by R1Snake