arkhipenko/TaskScheduler #120: Do you have an example about pulseIn() function?

jatu-studiobox, 2021-05-07:
Hi,

I have some problem about using TaskSchedule with pulseIn() function. 

Below code is normal process. I cannot imagine apply TaskSchedule with it. It keep pulseIn() LOW duration signal every loop() function, until 5 minutes, then it calculate AQI value from keep  pulseIn() LOW duration. 

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(DUST_SENSOR_DIGITAL_PIN_PM10, INPUT);
  pinMode(DUST_SENSOR_DIGITAL_PIN_PM25, INPUT);

  // wait 60s for DSM501 to warm up
  for (int i = 1; i <= 60; i++)
  {
    delay(1000); // 1s
    Serial.print(i);
    Serial.println(" s (wait 60s for DSM501 to warm up)");
  }

  Serial.println("Ready!");

  AQI.starttime = millis();
  
  timer.setInterval(sampletime_ms);
}

void loop() {
  AQI.lowpulseoccupancyPM10 += pulseIn(DUST_SENSOR_DIGITAL_PIN_PM10, LOW);
  AQI.lowpulseoccupancyPM25 += pulseIn(DUST_SENSOR_DIGITAL_PIN_PM25, LOW);

  if (timer.isReady()) {
    updateAQI();
    timer.reset();
  }
}

refer: https://diyprojects.io/calculate-air-quality-index-iaq-iqa-dsm501-arduino-esp8266

Please advice me.

Thank you very much

arkhipenko, 2021-05-07:
`pulseIn()` is a blocking function (similar to `delay()`) and could NOT be used with TaskScheduler without affecting the scheduling. 
You have to use pin interrupts instead if you would like to continue using cooperative multitasking. 

An implementation of such for ultrasound distance sensing was done here:
https://github.com/arkhipenko/pumpkin/blob/master/Pumpkin_Firmware.ino#L296

closed 2021-05-07 by arkhipenko

jatu-studiobox, 2021-05-12:
Hi arkhipenko,

Thank you very much for your advise. 

Should I use interupts for keeping duration, instead of pulseIn function?

arkhipenko, 2021-05-13:
You can always use pin change interrupts to detect start time. There is no need to use a blocking function that holds all other processes.

jatu-studiobox, 2021-06-01:
Hi arkhipenko,

Thank you very much for your advise. Now I can use interrupt and not use blocking function, pulseIn() function.

But my next problem is, the summary result duration (from my task and interrupt for gathering LOW pulse duration) does not same value which is from pulseIn() function. 

I am not sure, I must tune task INTERVAL time, so that the summary duration value is the same as pulseIn() function, or not?