arkhipenko/TaskScheduler #121: 'StatusRequest' was not declared in this scope   compile error

noweare1, 2021-05-16:
Hello, I am learning to use your library. Using the arduino ide, I get the above error when compiling the example Scheduler_example00_Blink.ino  which is renamed to mult_task1.ino   

Any help is appreciated.

I used git clone to place your repo of files in my libraries directory and #include <TaskScheduler.h> at the top of my file.

Here is the esp32 board I am using :
Arduino: 1.8.9 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Here are the error messages:
multi_task1:222:3: error: 'StatusRequest' was not declared in this scope
   StatusRequest* r = tBlink4On.getInternalStatusRequest();
   ^
multi_task1:222:18: error: 'r' was not declared in this scope
   StatusRequest* r = tBlink4On.getInternalStatusRequest();
                  ^
multi_task1:222:32: error: 'class Task' has no member named 'getInternalStatusRequest'
   StatusRequest* r = tBlink4On.getInternalStatusRequest();
                                ^
multi_task1:223:14: error: 'class Task' has no member named 'waitForDelayed'
   tBlink4Off.waitForDelayed( r );
              ^
/home/jrm/Documents/arduino/multi_task1/multi_task1.ino: In function 'void blink42()':
multi_task1:231:3: error: 'StatusRequest' was not declared in this scope
   StatusRequest* r = tBlink4Off.getInternalStatusRequest();
   ^
multi_task1:231:18: error: 'r' was not declared in this scope
   StatusRequest* r = tBlink4Off.getInternalStatusRequest();
                  ^
multi_task1:231:33: error: 'class Task' has no member named 'getInternalStatusRequest'
   StatusRequest* r = tBlink4Off.getInternalStatusRequest();
                                 ^
multi_task1:232:13: error: 'class Task' has no member named 'waitForDelayed'
   tBlink4On.waitForDelayed( r );
             ^
Using library TaskScheduler at version 3.2.2 in folder: /home/jrm/Documents/arduino/libraries/TaskScheduler 
exit status 1
'StatusRequest' was not declared in this scope

arkhipenko, 2021-05-17:
You didn't enable the status request functionality.
Please review the WiKi with respect to compile options

```
// #define _TASK_TIMECRITICAL      // Enable monitoring scheduling overruns
#define _TASK_SLEEP_ON_IDLE_RUN // Enable 1 ms SLEEP_IDLE powerdowns between tasks if no callback methods were invoked during the pass
#define _TASK_STATUS_REQUEST    // Compile with support for StatusRequest functionality - triggering tasks on status change events in addition to time only
// #define _TASK_WDT_IDS           // Compile with support for wdt control points and task ids
// #define _TASK_LTS_POINTER       // Compile with support for local task storage pointer
// #define _TASK_PRIORITY          // Support for layered scheduling priority
// #define _TASK_MICRO_RES         // Support for microsecond resolution
// #define _TASK_STD_FUNCTION      // Support for std::function (ESP8266 and ESP32 ONLY)
// #define _TASK_DEBUG             // Make all methods and variables public for debug purposes
// #define _TASK_INLINE            // Make all methods "inline" - needed to support some multi-tab, multi-file implementations
// #define _TASK_TIMEOUT           // Support for overall task timeout
// #define _TASK_OO_CALLBACKS      // Support for dynamic callback method binding
#include <TaskScheduler.h>
```

noweare1, 2021-05-17:
Thanks, I knew it had to be something stupid on my part. Thanks for the great software. Keeps me from having to learn
FreeRtos

arkhipenko, 2021-05-17:
FreeRTOS and TS could be used together! When you are ready - check this out. 
Good luck!

https://www.hackster.io/anatoli-arkhipenko/minecraft-interactive-do-not-enter-sword-sign-esp32-cam-cd1b07

Co-existance of Pre-emptive and Cooperative multi-tasking.

closed 2021-05-17 by arkhipenko