arkhipenko/TaskScheduler #122: passing Taskscheduler doesn't work

eleson2, 2021-05-21:
Code below works, but if I replace Task definition with the commented one, the callback isn't called.
Am I wrong in thinking that it should work?

`#include <Arduino.h> 
#include "TaskScheduler.h"  // V 3.2.2

void callback(){
  static int a = 0;
  digitalWrite(LED_BUILTIN,a);
  a=1-a;
}

Scheduler ts;
Task blink(1000,TASK_FOREVER,&callback); //,&ts
// Task blink(1000,TASK_FOREVER,&callback,&ts); 

void setup() {
  pinMode(LED_BUILTIN,OUTPUT);
  ts.init();
  ts.addTask(blink);
  blink.enable();
};

void loop() { ts.execute(); }; `

arkhipenko, 2021-05-21:
You need to decide how you want to assign tasks to the chain.
Either at the construction time:
```
Task blink(1000,TASK_FOREVER,&callback,&ts); 
```

or in the `setup()`:
```
Task blink(1000,TASK_FOREVER,&callback); //,&ts
...
  ts.addTask(blink);
...
```

Can not do both.
If you do both you are creating a task chain consisting of 2  tasks referencing itself - it's either a crash or a infinite loop somewhere as a result.

closed 2021-05-21 by arkhipenko

eleson2, 2021-05-21:
Thanks, makes perfect sense.

arkhipenko, 2021-05-21:
I just tested it on my Uno and it does work:

```
#include "TaskScheduler.h" 
void callback(){
  static int a = 0;
  digitalWrite(LED_BUILTIN,a&1);
  a++;
}

Scheduler ts;
Task blink(1000,TASK_FOREVER,&callback,&ts); 

void setup() {
  pinMode(LED_BUILTIN,OUTPUT);
//  ts.init();
  blink.enable();
}

void loop() { 
   ts.execute(); 
}
```

arkhipenko, 2021-05-21:
Do not init the scheduler if you assign it to the Task at creation time. That was the problem.
The scheduler was init'ed at the Task creation time. 

https://github.com/arkhipenko/TaskScheduler/wiki/API-Task-Scheduler#void-init