arkhipenko/TaskScheduler #124: Virtual destructor on bare metal

kevinpt, 2021-07-11:
The switch to a virtual destructor for the Task class in #86 is problematic outside the Arduino environment. The current code forces the creation of a vtable for Task and needs operator delete to exist. There is no need for a virtual destructor if none of the other methods are virtual. Composition can achieve the same result as subclassing. It should be conditional on the definition of _TASK_OO_CALLBACKS.

TD-er, 2021-07-11:
You do also need a virtual destructor if another class inherits from your class.
For example if you have a pointer to the base class and destruct it, you really need a virtual destructor.
This is not depending on the presence of virtual functions.

So I really wonder in what use case this causes an issue.

arkhipenko, 2021-07-30:
@kevinpt: I have reviewed the interface and I am not sure that any of the other methods are actually meant to be virtual - since they represent fundamental task behaviors not meant to be changed. 
If you disagree - please provide a use case proving otherwise. 
However - a virtual destructor is indeed needed to properly destroy inherited objects.

closed 2021-07-30 by arkhipenko