arkhipenko/TaskScheduler #131: Feature request: Add optional name field to tasks

joemcool, 2021-11-14:
I'd like to add the option to specify a name for a Task.  I'm imagining this could be set when the constructor is run, and reset later with an accessor method.  I'm planning to use this feature to generate a list of currently active tasks with some information about each (interval, active/disabled status, overrun status, and eventually per Task load).

Is there any objection to putting this feature in?  I didn't see any old issues discussing it, and was a little surprised it wasn't there already.  Maybe there's a reason it was not put in?

arkhipenko, 2021-11-14:
The reason why it was never implemented (and will never be) is that I strive to minimize the memory footprint of the library. Task names are non-essential information. You are free and welcome to maintain a map of task IDs to a an array of names if you really need this functionality. It has never been before requested and I personally never needed such. 
I know that FreeRTOS assigns a name to a task, and this is in my opinion just a waste of memory. 

You can identify Tasks by IDs if you enable _TASK_WDT_IDS

joemcool, 2021-11-14:
Got it.  Thanks for the context!

closed 2021-11-14 by joemcool

arkhipenko, 2021-11-14:
You are welcome, and thank you for doing the research on the previous issues before requesting - this is rare and appreciated.