arkhipenko/TaskScheduler #132: 'Scheduler' does not name a type error for nRF5240 with Arduino Core

viggi1000, 2021-12-02:
Hi, 

I have been receiving type error while running any of the examples with the nRF5240 controller running the [arduino core](https://github.com/adafruit/Adafruit_nRF52_Arduino). While this library states that it has been tested with nRF52832 controller, I am not sure if this error is due to the different model number?

Here is the compile error from arduino:

```
:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:139:10: error: 'Scheduler' does not name a type
  139 |   extern Scheduler* iSleepScheduler;
      |          ^~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:165:93: error: 'Scheduler' is not a type
  165 |     INLINE Task(unsigned long aInterval=0, long aIterations=0, TaskCallback aCallback=NULL, Scheduler* aScheduler=NULL, bool aEnable=false, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);
      |                                                                                             ^~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:292:5: error: 'Scheduler' does not name a type
  292 |     Scheduler                *iScheduler;            // pointer to the current scheduler
      |     ^~~~~~~~~
In file included from C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\examples\Scheduler_example09_TimeCritical\Scheduler_example09_TimeCritical.ino:13:
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:254:3: error: 'Scheduler' does not name a type
  254 |   Scheduler* iSleepScheduler;
      |   ^~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:291:80: error: 'Scheduler' is not a type
  291 | Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {
      |                                                                                ^~~~~~~~~
In file included from C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\examples\Scheduler_example09_TimeCritical\Scheduler_example09_TimeCritical.ino:13:
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In constructor 'Task::Task(long unsigned int, long int, TaskCallback, int*, bool, TaskOnEnable, TaskOnDisable)':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:296:33: error: request for member 'addTask' in '* aScheduler', which is of non-class type 'int'
  296 |     if (aScheduler) aScheduler->addTask(*this);
      |                                 ^~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In destructor 'virtual Task::~Task()':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:311:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  311 |     if (iScheduler)
      |         ^~~~~~~~~~
      |         Scheduler
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Task::reset()':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:457:5: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  457 |     iScheduler = NULL;
      |     ^~~~~~~~~~
      |     Scheduler
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::enable()':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:550:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  550 |     if (iScheduler) { // activation without active scheduler does not make sense
      |         ^~~~~~~~~~
      |         Scheduler
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::disable()':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:685:25: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  685 |         Task *current = iScheduler->iCurrent;
      |                         ^~~~~~~~~~
      |                         Scheduler
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::addTask(Task&)':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:818:15: error: 'class Task' has no member named 'iScheduler'
  818 |     if (aTask.iScheduler != NULL)
      |               ^~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:821:11: error: 'class Task' has no member named 'iScheduler'
  821 |     aTask.iScheduler = this;
      |           ^~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::deleteTask(Task&)':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:842:15: error: 'class Task' has no member named 'iScheduler'
  842 |     if (aTask.iScheduler != this)
      |               ^~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:845:11: error: 'class Task' has no member named 'iScheduler'
  845 |     aTask.iScheduler = NULL;
      |           ^~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::setSleepMethod(SleepCallback)':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1006:9: error: 'iSleepScheduler' was not declared in this scope
 1006 |         iSleepScheduler = this;
      |         ^~~~~~~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Scheduler::execute()':
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1189:14: error: 'iSleepScheduler' was not declared in this scope
 1189 |         if ( iSleepScheduler == this ) { // only one scheduler should make the MC go to sleep.
      |              ^~~~~~~~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\examples\Scheduler_example09_TimeCritical\Scheduler_example09_TimeCritical.ino: At global scope:
Scheduler_example09_TimeCritical:22:1: error: 'Scheduler' does not name a type
   22 | Scheduler runner;
      | ^~~~~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\examples\Scheduler_example09_TimeCritical\Scheduler_example09_TimeCritical.ino: In function 'void setup()':
Scheduler_example09_TimeCritical:41:3: error: 'runner' was not declared in this scope
   41 |   runner.init();
      |   ^~~~~~
C:\Users\viggi\Documents\Arduino\libraries\TaskScheduler\examples\Scheduler_example09_TimeCritical\Scheduler_example09_TimeCritical.ino: In function 'void loop()':
Scheduler_example09_TimeCritical:56:3: error: 'runner' was not declared in this scope
   56 |   runner.execute();
      |   ^~~~~~
exit status 1
'Scheduler' does not name a type
```

arkhipenko, 2021-12-02:
Are you using Arduino IDE or PlatformIO?
Please send your source - I can't help you without it. You can DM me to arkhipenko@hotmail.com

yourapiexpert, 2022-01-23:
I can confirm the same.  It occurs with the basic examples in the examples folder so no special source code is required.

I am using PlatformIO.

arkhipenko, 2022-01-24:
I just changed example #27 to run with nRF52840-DK and it compiled without errors. That is why I need to see your Platform IO project to be able to tell where the issue is. 
[Scheduler_example27_PlatformIO.zip](https://github.com/arkhipenko/TaskScheduler/files/7922589/Scheduler_example27_PlatformIO.zip)

yourapiexpert, 2022-01-24:
Same source as your .zip, different target board but nothing different in platformio.ini that would make a difference.

However, digging into the matter it seems that the 'Scheduler' variable of TaskScheduler conflicts with the Scheduler variable of SchedulerRTOS from FreeRTOS given that this lies beneath the surface of this platform.

Maybe that helps?

arkhipenko, 2022-01-24:
The difference is that I don't get the error and I have used TS with FreeRTOS previously without conflicts...
I would appreciate if you send the versions of your PlatformIO components and a zipped project that fails. 
arkhipenko@hotmail.com

yourapiexpert, 2022-01-24:
No, sorry, this is the best I can do.  My project is extremely sensitive and I have moved off this scheduler due to time constraints.

arkhipenko, 2022-01-24:
OK. Sorry about that. I cannot fix what I cannot reproduce. Good luck with your project!

closed 2022-01-24 by arkhipenko

yourapiexpert, 2022-01-24:
Right...here's the real culprit for what it's worth:

https://www.arduino.cc/en/Reference/Scheduler

This is enabled/compiled so not sure how to turn it off.  However, it seems a localized problem.

arkhipenko, 2022-01-25:
It's not RTOS. It's a separate scheduler that should not be used with TS. Of course there is a conflict.