arkhipenko/TaskScheduler #133: Compiler warning: 'uint32_t _task_micros()'  defined but not used

numericOverflow, 2021-12-31:
When using this library using Visual Studio Code, the compiler throws a warning about one of the functions:

`Compiling .pio\build\nodemcuv2\src\main.master.cpp.o
In file included from src\main.master.cpp:117:
.pio\libdeps\nodemcuv2\TaskScheduler\src/TaskScheduler.h:280:17: warning: 'uint32_t _task_micros()' 
defined but not used [-Wunused-function]
280 | static uint32_t _task_micros() {return micros();}
      |                 ^~~~~~~~~~~~`

[image]

It looks like there's a [PR to fix](https://github.com/arkhipenko/TaskScheduler/pull/129), can you look into fixing this issue?

I'm using the latest TaskScheduler lib version as of today (v3.6.0)

andriy063, 2023-01-02:
The same issue

abrahmx, 2023-02-24:
Same issue here 
[image]

arkhipenko, 2023-09-27:
This is fixed as of 3.8.2

closed 2023-09-27 by arkhipenko

lexskal, 2024-04-25:
// ------------------ TaskScheduler implementation --------------------

#ifndef _TASK_EXTERNAL_TIME
static uint32_t _task_millis() {return millis();}
//static uint32_t _task_micros() {return micros();}
#endif  //  _TASK_EXTERNAL_TIME