arkhipenko/TaskScheduler #135: How to get 'untilInterval'?

vatsake, 2022-01-30:
I can get the interval with `getInterval()`, but is there a way to see time UNTIL the task runs? or calculate?

arkhipenko, 2022-01-30:
I am not sure I understand the question. 
Do you want to know how much time until the Task will be invoked?
You can do that from the scheduler:
https://github.com/arkhipenko/TaskScheduler/wiki/API-Task-Scheduler#long-timeuntilnextiteration-task-atask

If you want to know how long the task is going to run - that depends on the callback code and cannot be estimated as the scheduler does not preempt - you have to return from the callback.

arkhipenko, 2022-01-30:
You also need to appreciate the invocation is not guaranteed at that time - it depends on whether all the tasks in front of this one will behave and exit quickly for the schedule to hold.

closed 2022-01-30 by arkhipenko