arkhipenko/TaskScheduler #136: TaskScheduler conflicts with ArduinoOTA on ESP32

vindolin, 2022-02-17:
I tried the library on an ESP8266 and it worked perfectly.
When I tried to migrate my program to an ESP32 it crashed with the following error:

`assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)`

It took me some time to pinpoint the problem and it seems to somehow conflict  with the ArduinoOTA library.
When I commented out the OTA lines the problem was gone.
The exact same program runs perfectly on an ESP8266.

```cpp
#include <TaskScheduler.h>
#include <ArduinoOTA.h>

uint16_t counter = 0;

Scheduler runner;
void t1Callback();

Task t1(1000, TASK_FOREVER, &t1Callback);

void t1Callback() {
    Serial.println(++counter);
}

void setup() {
    ArduinoOTA.setHostname("deadmeat");
    ArduinoOTA.begin();
    Serial.begin(115200);

    runner.init();
    runner.addTask(t1);

    t1.enable();
}

void loop() {
    ArduinoOTA.handle();
    runner.execute();
}
```

vindolin, 2022-02-17:
I might have found the problem and will take the issue offline for the moment.

closed 2022-02-17 by vindolin

vindolin, 2022-02-17:
Ok found it.

I migrated my own IOT helper library from the Ticker lib to TaskScheduler and must have overlooked a line where I call a setupWifi function before the setupOTA one.
I thought I did the same thing with an enabled task but that one get's called the first time when runner.execute runs.
All good now 😅

arkhipenko, 2022-02-17:
I also recommend moving everything else for the `loop` to `Tasks`.
So `ArduinoOTA.handle();` runs on a more reasonable schedule.

vindolin, 2022-02-19:
Yep and thanks for the great library!