arkhipenko/TaskScheduler #140: Adafruit bootloader : 'Scheduler does not name a type'

nicolascolsoul, 2022-03-18:
Hello,

I usually use your TaskScheduler on Arduino with great pleasure. Thanks for that.

Today I flashed an Adafruit Feather bootloader on a nrf52840 custom board, it works fine. When I import the Scheduler library and want to use it, I get an error that appears and I don't know why. Do you have an idea ? It seems to me that nrf52 is supported. 

Code Snippet :
`
#include <TaskScheduler.h>
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>

Scheduler runner;
Task t1(100, TASK_FOREVER, &ble_task);
Task t2(1000, TASK_FOREVER, &battery_monitoring_task);

void setup() 
{
  runner.addTask(t1);
  runner.addTask(t2);

  t1.enable();
  t2.enable();
}
void loop() 
{
  runner.execute();
}
`

And the error I get is 'Scheduler' does not name a type.
`
In file included from C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:225,
                 from C:\Users\Nicolas Colsoul\Documents\Arduino\pinout_test\pinout_test.ino:1:
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:165:93: error: 'Scheduler' is not a type
  165 |     INLINE Task(unsigned long aInterval=0, long aIterations=0, TaskCallback aCallback=NULL, Scheduler* aScheduler=NULL, bool aEnable=false, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);
      |                                                                                             ^~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:293:5: error: 'Scheduler' does not name a type
  293 |     Scheduler                *iScheduler;            // pointer to the current scheduler
      |     ^~~~~~~~~
In file included from C:\Users\Nicolas Colsoul\Documents\Arduino\pinout_test\pinout_test.ino:1:
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:303:80: error: 'Scheduler' is not a type
  303 | Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {
      |                                                                                ^~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In constructor 'Task::Task(long unsigned int, long int, TaskCallback, int*, bool, TaskOnEnable, TaskOnDisable)':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:308:33: error: request for member 'addTask' in '* aScheduler', which is of non-class type 'int'
  308 |     if (aScheduler) aScheduler->addTask(*this);
      |                                 ^~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In destructor 'virtual Task::~Task()':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:323:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  323 |     if (iScheduler)
      |         ^~~~~~~~~~
      |         Scheduler
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Task::reset()':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:498:5: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  498 |     iScheduler = NULL;
      |     ^~~~~~~~~~
      |     Scheduler
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::enable()':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:636:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  636 |     if (iScheduler) { // activation without active scheduler does not make sense
      |         ^~~~~~~~~~
      |         Scheduler
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::disable()':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:860:25: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  860 |         Task *current = iScheduler->iCurrent;
      |                         ^~~~~~~~~~
      |                         Scheduler
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::addTask(Task&)':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:995:15: error: 'class Task' has no member named 'iScheduler'
  995 |     if (aTask.iScheduler != NULL)
      |               ^~~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1000:11: error: 'class Task' has no member named 'iScheduler'
 1000 |     aTask.iScheduler = this;
      |           ^~~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::deleteTask(Task&)':
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1023:15: error: 'class Task' has no member named 'iScheduler'
 1023 |     if (aTask.iScheduler != this)
      |               ^~~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1028:11: error: 'class Task' has no member named 'iScheduler'
 1028 |     aTask.iScheduler = NULL;
      |           ^~~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\pinout_test\pinout_test.ino: At global scope:
pinout_test:30:1: error: 'Scheduler' does not name a type
   30 | Scheduler runner;
      | ^~~~~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\pinout_test\pinout_test.ino: In function 'void setup()':
pinout_test:55:3: error: 'runner' was not declared in this scope
   55 |   runner.addTask(t1);
      |   ^~~~~~
C:\Users\Nicolas Colsoul\Documents\Arduino\pinout_test\pinout_test.ino: In function 'void loop()':
pinout_test:76:3: error: 'runner' was not declared in this scope
   76 |   runner.execute();
      |   ^~~~~~
exit status 1
'Scheduler' does not name a type
`

If you have an  idea I would be very grateful. Thanks in advance.

arkhipenko, 2022-03-18:
I suspect nRF52840 Adafruit library uses their own scheduler and that creates name conflict. I have not looked into it unfortunately

hank870907, 2022-04-28:
Yeah, @arkhipenko is right.
Just change all the names of the "Scheduler" so something else and you are good to go.
Here's a copy of the library with the Scheduler changed to SchedulerAUCO.
[TaskScheduler.zip](https://github.com/arkhipenko/TaskScheduler/files/8587260/TaskScheduler.zip)
.

closed 2022-07-08 by arkhipenko