arkhipenko/TaskScheduler #143: #include error from c++ compiler due to multiple definition error

theionSami, 2022-05-18:
first i want to thank you for the great work.
As the cpp compiler seems not happy only with header files, it complains whenever i include the "taskScheduler.h" in other header than main.cpp
As far i know it need to find the implementation files for your library. otherwise it complains about multiple definitinons ...
Is there any version/ branch where we can find .h and .cpp from yours.
Thank you

arkhipenko, 2022-05-18:
This is how you do it:

https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example27_PlatformIO
[https://opengraph.githubassets.com/6b9ea2568b35c75afe9cd1a947da50e9dd93fb6b083d59db82f546f27a4511a8/arkhipenko/TaskScheduler]<https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example27_PlatformIO>
TaskScheduler/examples/Scheduler_example27_PlatformIO at master · arkhipenko/TaskScheduler<https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example27_PlatformIO>
Cooperative multitasking for Arduino, ESPx, STM32, nRF and other microcontrollers - TaskScheduler/examples/Scheduler_example27_PlatformIO at master · arkhipenko/TaskScheduler
github.com

closed 2022-05-18 by arkhipenko

Jozott00, 2022-07-08:
For everyone who also has the linking problem with `TaskScheduler.h`:

It is important that only one file with the `#define TaskScheduler.h` macro is compiled! Therefore the PlatformIO example has a `TaskScheduler.cpp` file with just this macro. 

All other files (including `main.cpp`) that have to use the library must use `#define TaskSchedulerDeclarations.h` to get linked without an error!