arkhipenko/TaskScheduler #145: Cannot compile blink LED example

edwardCao, 2022-10-02:
Hi!

I am a Arduino newbie trying out the library but cannot compile the LED blinking example in example folder. 

Arduino 2.0.0
Board: Adafruit Feather nRF52840 Sense

Error message:
In file included from c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:225,
                 from C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:26:
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:140:10: error: 'Scheduler' does not name a type
  140 |   extern Scheduler* iSleepScheduler;
      |          ^~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:165:93: error: 'Scheduler' is not a type
  165 |     INLINE Task(unsigned long aInterval=0, long aIterations=0, TaskCallback aCallback=NULL, Scheduler* aScheduler=NULL, bool aEnable=false, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);
      |                                                                                             ^~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:175:41: error: 'Scheduler' is not a type
  175 |     INLINE Task(TaskCallback aCallback, Scheduler* aScheduler, TaskOnEnable aOnEnable=NULL, TaskOnDisable aOnDisable=NULL);
      |                                         ^~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskSchedulerDeclarations.h:293:5: error: 'Scheduler' does not name a type
  293 |     Scheduler                *iScheduler;            // pointer to the current scheduler
      |     ^~~~~~~~~
In file included from C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:26:
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:266:3: error: 'Scheduler' does not name a type
  266 |   Scheduler* iSleepScheduler;
      |   ^~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:303:80: error: 'Scheduler' is not a type
  303 | Task::Task( unsigned long aInterval, long aIterations, TaskCallback aCallback, Scheduler* aScheduler, bool aEnable, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {
      |                                                                                ^~~~~~~~~
In file included from C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:26:
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In constructor 'Task::Task(long unsigned int, long int, TaskCallback, int*, bool, TaskOnEnable, TaskOnDisable)':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:308:33: error: request for member 'addTask' in '* aScheduler', which is of non-class type 'int'
  308 |     if (aScheduler) aScheduler->addTask(*this);
      |                                 ^~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In destructor 'virtual Task::~Task()':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:323:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  323 |     if (iScheduler)
      |         ^~~~~~~~~~
      |         Scheduler
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: At global scope:
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:340:37: error: 'Scheduler' is not a type
  340 | Task::Task( TaskCallback aCallback, Scheduler* aScheduler, TaskOnEnable aOnEnable, TaskOnDisable aOnDisable ) {
      |                                     ^~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In constructor 'Task::Task(TaskCallback, int*, TaskOnEnable, TaskOnDisable)':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:345:33: error: request for member 'addTask' in '* aScheduler', which is of non-class type 'int'
  345 |     if (aScheduler) aScheduler->addTask(*this);
      |                                 ^~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Task::reset()':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:498:5: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  498 |     iScheduler = NULL;
      |     ^~~~~~~~~~
      |     Scheduler
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::enable()':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:636:9: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  636 |     if (iScheduler) { // activation without active scheduler does not make sense
      |         ^~~~~~~~~~
      |         Scheduler
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'bool Task::disable()':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:860:25: error: 'iScheduler' was not declared in this scope; did you mean 'Scheduler'?
  860 |         Task *current = iScheduler->iCurrent;
      |                         ^~~~~~~~~~
      |                         Scheduler
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::addTask(Task&)':
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:995:15: error: 'class Task' has no member named 'iScheduler'
  995 |     if (aTask.iScheduler != NULL)
      |               ^~~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:1000:11: error: 'class Task' has no member named 'iScheduler'
 1000 |     aTask.iScheduler = this;
      |           ^~~~~~~~~~
c:\Users\myuser\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Scheduler::deleteTask(Task&)':
[... 25 lines cut ...]
   69 | Task tBlink3 (PERIOD3 * TASK_MILLISECOND, DURATION / PERIOD3, &blink3CB, &ts, false);
      |                                                                           ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:79:68: error: 'ts' was not declared in this scope
   79 | Task tBlink4On  ( PERIOD4 * TASK_MILLISECOND, TASK_ONCE, blink41, &ts, false, &blink41OE );
      |                                                                    ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:80:68: error: 'ts' was not declared in this scope
   80 | Task tBlink4Off ( PERIOD4 * TASK_MILLISECOND, TASK_ONCE, blink42, &ts, false, NULL, &blink42OD );
      |                                                                    ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:91:74: error: 'ts' was not declared in this scope
   91 | Task tBlink5On  ( 600 * TASK_MILLISECOND, DURATION / PERIOD5, &blink51, &ts, false, &blink51OE );
      |                                                                          ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:92:74: error: 'ts' was not declared in this scope
   92 | Task tBlink5Off ( 600 * TASK_MILLISECOND, DURATION / PERIOD5, &blink52, &ts, false, NULL, &blink52OD );
      |                                                                          ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:102:76: error: 'ts' was not declared in this scope
  102 | Task tBlink6 ( PERIOD6 * TASK_MILLISECOND, DURATION / PERIOD6, &blink6CB, &ts, false, &blink6OE, &blink6OD );
      |                                                                            ^~
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino: In function 'void loop()':
C:\Users\myuser\Documents\Arduino\sketch_blink\sketch_blink.ino:117:3: error: 'ts' was not declared in this scope
  117 |   ts.execute();
      |   ^~

exit status 1

Compilation error: 'Scheduler' does not name a type

arkhipenko, 2022-10-04:
hi, @edwardCao - 
I confirm that I can reproduce your issue using Adafruit nFR50 Arduino Core. 
I do not understand why it behaves this way while on other boards same code compiles without issues. 
I will have to have a look into it further.

arkhipenko, 2022-10-04:
I am afraid this is because Adafruit core uses FreeRTOS and there is a forward declaration of a Scheduler, which then conflicts with my class definition of a Scheduler. Name conflict - I do not know what to do about it since changing the class name will instantly make all previous code fail to compile. I need to think if there is a way to make this change backwards-compatible.

arkhipenko, 2022-10-04:
Yes, there is an object called Scheduler in rtos.cpp and rtos.h files in Adafruit nRF core.
Both Adafruit developer and myself did not think of a more unique name, now here we are...

edwardCao, 2022-10-04:
@arkhipenko thanks for looking into this issue and thanks to @vortigont for providing a solution/patch. I will try this solution on user side (my knowledge about the IDE is limited to using only one sketch file :-) ).

arkhipenko, 2022-10-10:
Fixed by 3.6.2 (in testing branch for now)

closed 2022-10-10 by arkhipenko