arkhipenko/TaskScheduler #154: ESP32: WD resets MCU if SLEEP feature is enabled

vortigont, 2022-10-25:
I've hit the issue where watch dog triggers MCU reset if _TASK_SLEEP_ON_IDLE_RUN feature is used.
It came out of nowhere in one of my existing projects that was working before. So looks like some combination of Tasks periods triggers it. Reproducible at esp32 and esp32-s2, do not have other boards to test for now. Removing `_TASK_SLEEP_ON_IDLE_RUN` solves the issue completely.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

I'll try to isolate it and provide the RCA a bit later.

arkhipenko, 2022-10-25:
@vortigont - I would not use `_TASK_SLEEP_ON_IDLE_RUN  `on esp32 at all. It never worked reliably.

vortigont, 2022-10-26:
yeah, it does not work as expected. The thing is looks like you've merged this feature in 3.6.1 recently while before that it was a noop for esp32, that was my surprise to hit a regression.
Googled for same issues, looks like some older wroom boards has this issue with RTC watchdog, no solution found.
As for S2, my bad, it is not crashing same way but also does not work as expected, looses STA connection, misses beacon frames, etc.
I've checked ESP-IDF guides, it has some automatic light sleep mode via RTOS TICKLESS_IDLE, but this option is not enabled in Arduino's core (sigh!). And manual light_sleep call requires some other features to be set, like wifi/gpio wake, etc... Maybe better to disable it as it was before or else replace with vTaskDelay()? :)

arkhipenko, 2022-10-26:
> Maybe better to disable it as it was before or else replace with vTaskDelay()? :) 

or not use _TASK_SLEEP_ON_IDLE_RUN on esp32 at all... :)

vortigont, 2022-10-26:
yep, done.
that is easy if you know what caused the issue :)

closed 2022-10-26 by vortigont