arkhipenko/TaskScheduler #157: ESP32 dual core usage

n1ckdm, 2022-11-20:
Is there anything in particular I should do to ensure that both cores of the ESP32 are utilised when using TaskScheduler, or is the handled by the library internally?

Or does the library only support single core?

arkhipenko, 2022-11-20:
TS is a cooperative multi-tasking library and therefore runs in a single thread, single core.

arkhipenko, 2022-11-20:
ESP32 runs on FreeRTOS adopted to multi-core architecture. Details here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html?highlight=freertos

I have been quite successful in combining the benefits of both frameworks. Example is here:
https://github.com/arkhipenko/dnd

closed 2022-11-20 by arkhipenko