arkhipenko/TaskScheduler #159: V3.7.0 kills ESP32 WiFi

AcuarioCat, 2022-11-28:
My project uses an ESP32 with the latest Arduino code and V3.7.0 of the TaskScheduler.

My project has been working without problem for several months using V3.6.0.
After updating to V3.7.0 it started suffering from WiFi disconnection shortly after the setup was completed and the loop started running (roughly 10 seconds after loop started).
[WiFi-event] event: 5  WiFi lost connection

After much digging I found it was due to runner.execute(). Commenting this out and the code ran with no disconnects.
I removed all tasks from the code but the problem still occurred.
I then reverted back to V3.6.0 and the problem went away - no more disconnects. 

To double check I reinstalled V3.7.0 and the disconnects returned. Back to V3.6.0 and all is well again.

arkhipenko, 2022-11-29:
hI @AcuarioCat - do you, by any chance, use `_TASK_SLEEP_ON_IDLE_RUN` compile option?
If you do, please comment it out - it does not really work well on ESP32.

AcuarioCat, 2022-11-30:
Yes it was set in the old ts.hpp file.
I commented it out and now it seems to be working ok.
Maybe worth adding a note in the readme file?

closed 2022-11-30 by AcuarioCat