arkhipenko/TaskScheduler #160: Scheduler not executing on ESP8266

creahard, 2022-12-21:
I'm trying to use this library on an ESP8266 which I have used many times before, but I am migrating to PlatformIO vs Arduino IDE. Compiling and running the code below in the Arduino IDE with ESP8266 Core 2.7.4 runs correctly. I haven't been able to get ANY code using Core 3.0.2 to boot without an endless loop of exceptions and rebooting. Using PlatformIO and espressif8266 Core 3.0.2 it compiles and boots but the execute method of the Scheduler doesn't appear to run.

My main question is has version 3.7.0 of TaskScheduler been tested on an ESP8266 built with the latest espressif core framework?

+++++++++++test.ino / main.cpp+++++++++++++
#include <Arduino.h>
#include <TaskScheduler.h>

Scheduler runner;

void blinkSerial(void) {
  Serial.println("+");
}

Task serialBlink(1000, TASK_FOREVER, blinkSerial);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Serial.println("Setting up task scheduler.");
  runner.init();
  Serial.println("Adding serialBlink task.");
  runner.addTask(serialBlink);
  Serial.println("Setup is complete.");
}

void loop() {
  // put your main code here, to run repeatedly:
  runner.execute();
}

arkhipenko, 2022-12-22:
I think you are missing
serialBlink.enable();
at the end of setup()

Could you try with that added please?

Thanks

Apologies for autocorrect.

creahard, 2022-12-22:
Well, that was embarrassing. That was the missing piece. When I took a quick read through the larger source code I'm porting over I completely missed the enable method on some of the tasks (others get restartDelayed).

My apologies. Thank you very much for the quick reply!

closed 2022-12-22 by creahard

arkhipenko, 2022-12-22:
No worries. Happens to the best of us.

Apologies for autocorrect.