arkhipenko/TaskScheduler #164: Set the core where a task execute

sebarch, 2023-04-22:
Hi, it´s possible to set the core where a task is executed without using freertos? Thanks

arkhipenko, 2023-04-22:
Hi.

No. TaskScheduler is a cooperative library, so it run in one thread in whatever task the scheduler and task objects are being called from. In fact, if you do use multiple rtos tasks, you should compile with thread protection.

Having said that, you can influence which core the rtos tasks is running on.
For example and inspiration, please check the main sketch of the repo below.

https://github.com/arkhipenko/dnd/blob/master/esp32_dnd_rtos_mjpeg_multi/esp32_dnd_rtos_mjpeg_multi.ino

The method you'd want to look closely at is
creareCoreTasks()

Notice how the video streaming task and TaskScheduler task are created pinned to different cores.
Works like a charm.

Hope this helps.

Cheers,
Anatoli

Apologies for autocorrect.

sebarch, 2023-04-22:
Thanks for your quick replay

closed 2023-04-22 by sebarch