arkhipenko/TaskScheduler #169: Method to get the name of the Callback registered in the Task

georgevbsantiago, 2023-07-10:
Hello @arkhipenko 

First of all, THANK YOU SO MUCH for this fantastic library!!!

A doubt. Is there a method to get the current callback name of the task?
In this case, when we use Task as a finite state machine, it would be interesting to know which Callback is registered in Task.

Ex:
```
void function_01 () { ... }

void function_02 () { ... }

void function_03 () { ... }

task.setCallback(&function_01);
...
task.setCallback(&function_02);
....
task.setCallback(&function_03);
....
    if( equal(task.getCallback(), &function_02) ) {
    
    ....
    }
```

arkhipenko, 2023-07-12:
I never needed this functionality and it was not asked for until now, hence it does not exist in the current version.

georgevbsantiago, 2023-07-12:
I'm using Task as a finite state machine to manage the messages on the 20x04 Display.
The need to know which Callback is registered in the Task arose from the need to know which message was being viewed on the Display.
If a 'getCallback' method is made available in a future version, it would be possible to identify the current Callback registered in the Task.

Thanks.

arkhipenko, 2023-08-22:
> If a 'getCallback' method is made available in a future version, it would be possible to identify the current Callback registered in the Task.

The callback is a piece of code, not an object, so it can not really hold a "name". I can have Tasks return a pointer to the current callbacks, and you would need to compare to known pointers to figure out which one is assigned. 
Would that satisfy your requirement?

georgevbsantiago, 2023-08-22:
Hi, I believe so.

What would be the proper approach to purchasing the 'getCallback' method with the compared function pointer?
Something like? : `task.getCallback() == &function_02`

arkhipenko, 2023-08-23:
Yes, something like that I guess.

georgevbsantiago, 2023-08-23:
I'm a beginner in C++, but maybe this post can help:
https://stackoverflow.com/questions/12898227/comparing-function-pointers

arkhipenko, 2024-10-07:
Meanwhile you can use 
`void setLtsPointer(void *aPtr);`
and
`void *getLtsPointer();`
to store and compare the address of the callback method:

```
void function_01 () { ... }
task.setCallback(&function_01);
task.setLtsPointer((void*) &function_01);
```
would that work as a workaround?

(to enable LTS - use `#define _TASK_LTS_POINTER` compile option

closed 2024-10-07 by arkhipenko