arkhipenko/TaskScheduler #170: Async Webserver problem

sebosfato, 2023-08-19:
Hello im trying to implemente the async webserver along with the task scheduller 

but i seem to have find some exceptions i guess its because it run on background 

is there a way to set a task for it?

thanks

arkhipenko, 2023-08-19:
Post your code here, I will have a look.

sebosfato, 2023-08-19:
I believe the part that is giving problem is this.. 

String getContentType(String filename) {
  if (filename.endsWith(".htm") || filename.endsWith(".html")) return "text/html";
  else if (filename.endsWith(".css")) return "text/css";
  else if (filename.endsWith(".js")) return "application/javascript";
  else if (filename.endsWith(".png")) return "image/png";
  else if (filename.endsWith(".jpg") || filename.endsWith(".jpeg")) return "image/jpeg";
  else if (filename.endsWith(".gif")) return "image/gif";
  else if (filename.endsWith(".ico")) return "image/x-icon";
  else if (filename.endsWith(".xml")) return "text/xml";
  else if (filename.endsWith(".pdf")) return "application/pdf";
  else if (filename.endsWith(".zip")) return "application/zip";
  else if (filename.endsWith(".gz")) return "application/gzip";
  else if (filename.endsWith(".map")) return "application/javascript"; // You can add more file types here
  else if (filename.endsWith(".txt")) return "text/plain"; // For example, add .txt
  else {
    // Default content type if none of the known file types match
    return "application/octet-stream";
  }
}
bool handleFileRead(AsyncWebServerRequest *request) {
    String path = request->url();

    if (path.endsWith("/")) {
        path += "index.html";
    }

    String contentType = getContentType(path);
    String pathWithGz = path + ".gz";
    
    if (SPIFFS.exists(pathWithGz) || SPIFFS.exists(path)) {
       if (SPIFFS.exists(pathWithGz)) {
      path += ".gz";
    }
     
   

    File file = SPIFFS.open(path, "r");
    request->send(SPIFFS, path, contentType);
    file.close();
    return true;
    }

  if (path.endsWith("&end=end")){
    String path = request->url();
    ////Serial.println(path);
    //Serial.println(path.substring(5,13)); // extrai so a data da string do formulario
    datmes=path.substring(10,12).toInt();     //mes
    datano=path.substring(5,9).toInt();   //ano
    dat=path.substring(13,15).toInt();    //dia
      request->send(200);
  }
  
  I use this to make the content type etc like the old FS server… 
  
  
  
void handleNotFound(AsyncWebServerRequest *request) {
  if (!handleFileRead(request)) {
    if (captivePortal(request)) { // If captive portal, redirect instead of displaying the error page.
      return;
      //Serial.println("handle not found");
    }

    String message = F("File Not Found\n\n");
    message += F("URI: ");
    message += request->url();
    message += F("\nMethod: ");
    message += (request->method() == HTTP_GET) ? "GET" : "POST";
[... 389 lines cut ...]
18:46:39.041 -> 3ffffe30:  3ffeb5cc 2c9f0300 4000050c 3fffc278  
18:46:39.041 -> 3ffffe40:  401027d0 3fffc200 00000022 00000000  
18:46:39.076 -> 3ffffe50:  40100414 00000030 00000010 ffffffff  
18:46:39.076 -> 3ffffe60:  40100449 00000001 00000000 00000000  
18:46:39.076 -> 3ffffe70:  000186a0 0000d688 00000000 fffffffe  
18:46:39.076 -> 3ffffe80:  00000000 3fffc6fc 00000000 3fff066c  
18:46:39.076 -> 3ffffe90:  00000001 00000001 3fff067c 00000030  
18:46:39.076 -> 3ffffea0:  00000001 00000001 3fff067c 00000030  
18:46:39.076 -> 3ffffeb0:  00000001 00000001 00000000 00000030  
18:46:39.076 -> 3ffffec0:  40100f91 40100f8c 4023477c 00000000  
18:46:39.109 -> 3ffffed0:  00000000 00000000 00000000 fffffffe  
18:46:39.109 -> 3ffffee0:  ffffffff 3fffc6fc 00000001 3ffe85a8  
18:46:39.109 -> 3ffffef0:  00000000 3fffdad0 3fff3c74 00000030  
18:46:39.109 -> 3fffff00:  0000012c 3fffdad0 3fff3c74 00000030  
18:46:39.109 -> 3fffff10:  00000000 4bc6a7f0 00040cf2 005b8d80  
18:46:39.109 -> 3fffff20:  00000000 4bc6a7f0 00040cf2 3fff067c  
18:46:39.109 -> 3fffff30:  00000000 00000000 00000001 40100428  
18:46:39.147 -> 3fffff40:  00000000 00000001 3fff066c 00000001  
18:46:39.147 -> 3fffff50:  00000001 3fff067c 402346af 3fffefa0  
18:46:39.147 -> 3fffff60:  00000001 00000001 3fff066c 40208f3c  
18:46:39.147 -> 3fffff70:  00000000 80efeffe 4023f1d8 011cd9ac  
18:46:39.147 -> 3fffff80:  00000000 00000000 00000001 3fff3c74  
18:46:39.147 -> 3fffff90:  3fffdad0 00000000 3fff3c34 4020aeba  
18:46:39.147 -> 3fffffa0:  3fffdad0 00000000 3fff3c34 4023479c  
18:46:39.147 -> <<<stack<<<

arkhipenko, 2023-08-19:
First of all, please use the appropriate code quotes - it is difficult to read.
Secondly - where is the `setup()` method with scheduler init, adding tasks to the scheduler and activating them?
Thirdly - did you use the appropriate esp Exception Decoder tool to get the call stack trace of the exception? How do you know the problem is with the TaskScheduler and not other code?

sebosfato, 2023-08-19:
I tested the Async lib without the task scheduler and kind of worked fine.. .im seeing it give a lot of errors so im investigating what could be the cause… 

i imagine is because it may get busy in parallel with the tasks… 

here is the decoded :

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x4010063c: pvPortZalloc(size_t, char const*, int) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/heap.cpp line 301
0x402345f8: loop_task(ETSEvent*) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 205
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40246b75: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40246e02: new_linkoutput at glue-lwip/lwip-git.c line 260
0x40247224: ethernet_output at netif/ethernet.c line 312
0x40240bd8: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 193
0x402450c2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40240c05: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x402450c2: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40244ff8: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x40240d30: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x40240c05: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x402454f3: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 667
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40208e7c: Scheduler::execute() at /Users/devel/Documents/Arduino/libraries/TaskScheduler/src/TaskScheduler.h line 279
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4020aea8: loop() at /Users/devel/Desktop/Gardentoy Workspace /GardenToy/pro4sfinalasync/pro4sfinalasync.ino line 5321
0x4023479c: loop_wrapper() at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x4010042c: ets_post(uint8, ETSSignal, ETSParam) at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x402346af: __yield() at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_features.h line 92
0x40208f3c: Scheduler::execute() at /Users/devel/Documents/Arduino/libraries/TaskScheduler/src/TaskScheduler.h line 1054
0x4020aeba: loop() at /Users/devel/Desktop/Gardentoy Workspace /GardenToy/pro4sfinalasync/pro4sfinalasync.ino line 5325
0x4023479c: loop_wrapper() at /Users/devel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197

sebosfato, 2023-08-19:
The setup has this on it: its pretty much like from the original sketch

 server.on("/generate_204", handleRoot);
  server.on("/fwlink", handleRoot);  //Microsoft captive portal. Maybe not needed. Might be handled by notFound handler.
  server.onNotFound(handleNotFound);
  
  
WHen i try to call this it resets too:

void UPDATE() {
if (memoria==true){
  updatecounter++;
  if (updatecounter == 1 || updatecounter > 120){ 
    updatecounter=2;
  WiFiClient client;
  HTTPClient http;

  //First check version of update
  http.begin(client, "http://xxx/version.php");
  http.addHeader("Content-Type", "application/x-www-form-urlencoded");
  int httpCode = http.GET();            //Send the request

     Serial.println(httpCode);   //Print HTTP return code
  if (httpCode == 200 && (http.getString() != "") ) {
    versaoatual = http.getString();
    //versaoatual.trim();
    Serial.print("versaoatual ");
    Serial.println(versaoatual);
   // Serial.println("versao do update mais atual " + versaoatual);
  //  Serial.println("versao instalada " + String(versao));
  //  Serial.println ( "float versaoatual " + (String (versaoatual.toFloat())));
    http.end();
    
     ESPhttpUpdate.rebootOnUpdate(false);
      Serial.println("Update Sketch...");
      ESPhttpUpdate.setLedPin(LED, LOW);
      t_httpUpdate_return Ret = ESPhttpUpdate.update(client, "http://update.gardentoy.com.br/pro4sfinal/pro4sfinal.ino.bin");
      switch (Ret) {
        case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
          break;

        case HTTP_UPDATE_NO_UPDATES:
          Serial.println("HTTP_UPDATE_NO_UPDATES");
          break;

        case HTTP_UPDATE_OK:
          versao = versaofs;
          variavel = "versao";
          valor = versao;
          POSTDATA();
          digitalWrite(LED, LOW);
          Serial.println("HTTP_UPDATE_OK");
          Serial.println("Updated Sketch ooooookkk...");
          EEPROM.put(194, versao);  // long - so 8 bytes (next address would be '12')
          digitalWrite(LED, LOW);
          EEPROM.commit();
          Serial.println("HTTP_UPDATE_OK");
          Serial.println("versao atualizada" + String(versao));
          break;
         }     
          ESP.restart();
        } Serial.println("A versao era igual a fonte");
         } else {   Serial.println("verificar versao falhou ");
          }
  //http.end();
        }else {
      Serial.println("ja tentou update"); 
       }  }}

arkhipenko, 2023-08-19:
I still don't see where you add tasks to the scheduler and start them.
It could be the case that you are running execute() on an empty task chain, so it causes a null pointer exception.
It shouldn't, and it may be a bug, but you also should not run an empty task chain, right?

Apologies for autocorrect.

arkhipenko, 2023-08-19:
https://github.com/arkhipenko/TaskScheduler/blob/master/examples/Scheduler_example01/Scheduler_example01.ino#L64

This is what I don't see in your sketch.

arkhipenko, 2023-08-20:
I am afraid I will not be able to help you unless you send me full code. 
You send it to    a r k h i p e n k o @ h o t m a i l . c o m

sebosfato, 2023-08-20:
Im trying to make a resumed version of the code that gives the error and im going to send to you… thanks a lot! 

im trying to determine the problem and it may seem to be related to also other libraries so im trying to find out too.. 

I believe it may related to the scheduling of the tasks.. i heard the async webserver run on a task of its own so it could be interesting to find a way to associate with one of the tasks of the scheduler so it can decide when run or up to when it run… 

i noticed that the wifi manager and update and http requests also is not working so it may also have caused a conflict. 

I hope it may help others with same problem =)

arkhipenko, 2023-08-20:
OK.
TaskScheduler runs in one thread.
If you need to synchronize tasks from different threads and wait for async execution, it is better to use StatusRequest objects - as they are meant to be event-driven framework for TS.

https://github.com/arkhipenko/TaskScheduler/wiki/Full-Document#6-using-status-request-objects

https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example04_StatusRequest
https://github.com/arkhipenko/TaskScheduler/tree/master/examples/Scheduler_example05_StatusRequest

closed 2024-08-15 by arkhipenko