arkhipenko/TaskScheduler #171: esp32-c3 freezing after for tasks over 5800ms with _TASK_SLEEP_ON_IDLE_RUN. Regular esp32 works fine.

beachmiles, 2023-08-21:
It appears the use of _TASK_SLEEP_ON_IDLE_RUN with the esp32 is discouraged and it doesn't appear to reduce power usage much but maybe this will help with the issue?

//This works on the esp32-c3. Reg esp32 works.
Task t2(5800, TASK_FOREVER, &t2Callback);

//This works on the esp32-c3 for 2 runs then freezes. Reg esp32 works.
Task t2(5850, TASK_FOREVER, &t2Callback);

//This freezes the esp32-c3 after 1 run.. Reg esp32 works.
Task t2(5900, TASK_FOREVER, &t2Callback);

I verified this is the case with the real hardware as well as this slick simulator that also freezes in the same fashion.

The esp32-c3 example
https://wokwi.com/projects/373706302321604609

The regular esp32 same example.
https://wokwi.com/projects/373350195861220353

arkhipenko, 2023-08-22:
@beachmiles - there is a good reason why the use of `_TASK_SLEEP_ON_IDLE_RUN` is discouraged.
The current implementation (which may have an overflow bug indeed) uses light sleep mode which is incompatible with WiFi and BT. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes

You are more than welcome to play with the sleepMode implementation to find a better solution here:
https://github.com/arkhipenko/TaskScheduler/blob/4ccca1ac3d63a2b6fa853df3c93147fa4e8980da/src/TaskSchedulerSleepMethods.h#L48

but I would not use `_TASK_SLEEP_ON_IDLE_RUN`  if you use WiFi, definitely not on a single-core C3.

closed 2023-08-22 by arkhipenko

beachmiles, 2023-08-28:
Verified the old TaskScheduler 3.6.0 version does not freeze my esp32-c3 where 3.7.0 does freeze with the longer tasks. 
It looks like there was maybe no sleeping function set for esp32 with _TASK_SLEEP_ON_IDLE_RUN in this 3.6.0 version so that would explain it working unless Im not seeing the code correctly. 
I could have sworn I saw a lower power draw with _TASK_SLEEP_ON_IDLE_RUN being set on my esp32 though.
https://github.com/arkhipenko/TaskScheduler/compare/v3.6.0...v3.7.0