arkhipenko/TaskScheduler #188: C++ compiler warning: compound assignment with 'volatile'-qualified left operand is deprecated

GitMoDu, 2024-06-14:
C++20 deprecated compound assignment on volatiles. As Arduino cores update their C++ version, this warning will be emited on every compilation.

The Raspberry PI core already has addressed this issue: https://github.com/raspberrypi/pico-sdk/issues/1017

TD-er, 2024-06-14:
You should now use `std::atomic<int>` instead of `volatile int`.

GitMoDu, 2024-06-14:
The typical migration path I've seen is about re-structuring the operations, not to add the std:: dependency.

```
volatile a;
a += b;
```
turns into 
`a = a+b;`

TD-er, 2024-06-14:
See also discussions like these: https://stackoverflow.com/questions/59223814/why-is-volatile-deprecated-in-c20

vortigont, 2024-06-18:
think it would be easier to just remove `volatile` at all from this lib. Any way this poor-man's locks does not work as expected on, i.e. esp32 when concurrently adding/destructing task objects from Scheduler in a non-thread-safe manner because working with linked-list operations here is also not thread-safe.

arkhipenko, 2024-06-18:
> think it would be easier to just remove `volatile` at all from this lib. Any way this poor-man's locks does not work as expected on, i.e. esp32 when concurrently adding/destructing task objects from [...]

The whole point of the library is to be single-threaded cooperative multi-tasking. If you are multi-threading then RTOS is the answer.

arkhipenko, 2024-06-18:
The whole point of making a few variables `volatile` was the use of `restart()` , `restartDelayed()`, `signal()` and other methods from ISRs. That's the main use-case: in a pin interrupt routine restart a task or signal Status change. 
So what's the new safe way of making sure ISR-updated variables are reloaded in user-context?

arkhipenko, 2024-06-18:
v3.8.5: (testing branch)
2024-06-17:

updated volatile compound statements after C++20 deprecated compound assignment on volatiles

TD-er, 2024-06-18:
> Am I also reading it correctly that "This has been de-deprecated in C++23"? So nothing to do really?

Hmm that's new to me...
Got to read a bit more about this then as I wonder what the reasoning is (apart from "causing lots of work on older libraries/code")

vortigont, 2024-06-18:
> The whole point of the library is to be single-threaded cooperative multi-tasking. If you are multi-threading then RTOS is the answer.

true, scheduling and executing is single-threaded here, but other operations, like adding/removing/changing Tasks could be done from other threads, which is unsafe and `_TASK_THREAD_SAFE` does not always works as expected but only adds confusion. This is what I wanted to say here.
Usually I had to add guarded mutex in such cases, but it is  beyond the scope of this lib.

GitMoDu, 2024-06-18:
> Am I also reading it correctly that "This has been de-deprecated in C++23"? So nothing to do really?

I think it **was** un-deprecated (volatile keyword will stay), but not for the cases of volatile left-hand operators.

arkhipenko, 2024-06-18:
I updated the library, so we should be all good either way!Sent from a mobile device.Apologies for autocorrect.

arkhipenko, 2024-06-18:
You are right. `_TASK_THREAD_SAFE` was an attempt of thread safety for execution methods only (when enable or restart is executed in a separate thread, ts.execute() is running in another) I did not protect task creating and deleting as I very rarely do it actually. Usually tasks stay forever. Sent from a mobile device.Apologies for autocorrect.

vortigont, 2024-06-18:
@arkhipenko got it. THREAD_SAFE is just confusing define :)
I'm on the contrary use your scheduler to launch one-time or short living tasks. It's more convenient to spawn class instances with on heap allocation than defining/creating RTOS tasks and do stack size management .

arkhipenko, 2024-06-18:
Yes, much simpler! Don't need to worry about concurrency within the thread, and stack size is a killer, especially on esp32. I have very limited success with stack less than 2k! Sent from a mobile device.Apologies for autocorrect.

closed 2024-06-18 by arkhipenko