arkhipenko/TaskScheduler #190: Task.isEnabled() crashes ESP8266 when run in interrupt

CheatingBoy6890, 2024-08-12:
Hey, first off thanks for this great library, I personally really love it.
Sadly I have a problem. I've built a Lasertag program for the ESP8266 and I'm using TaskScheduler in it.
I have an ISR that gets triggered when a button ist pressed:

```cpp
void IRAM_ATTR onShoot()
{ 
  static volatile uint32_t debounce;
  if (millis() > debounce && alive && !TaskShoot.isEnabled())
  {
    TaskShoot.restartDelayed(weapons[myWeaponIndex].getTimeBeforeShot());
    debounce = millis() + 100;
  }
} 
```
The TaskCallback is the following:

```cpp
{
  if (millis() >= debounce_shoot && TaskReload.getRunCounter() < 1)
  {

    if (bullets > 0)
    {
      Serial.println(F("shooting"));
      sendMilesTag(myPlayerId, myTeamId, weapons[myWeaponIndex].getDammage());
      bullets--;
      pixels.setPixelColor(LED_SHOOTING, 0xFFFFFF);
      pixels.show();
      playAudio(SHOOT_SOUND);

      drawbullets(0, bullets);

      pixels.setPixelColor(LED_SHOOTING, 0);
      updatePixels();
    }
    else
    {
      playAudio(EMPTY_SOUND);
      // bullets = 0;
    }
    if (TaskShoot.getIterations() == 0 || bullets == 0) // starting the reloading when it's the last iteration of the task (for burst weapons) or the magazine is empty (for full auto)
    {

      Serial.println(F("starting Reload"));

      TaskReload.restartDelayed(weapons[myWeaponIndex].getTimeBeforeReload());
      if (TaskShoot.getIterations() == -1)
      {
        TaskShoot.setIterations(1);
      }
    }
    if (TaskShoot.isLastIteration())
    {
      debounce_shoot = millis() + weapons[myWeaponIndex].getTimebetwennshots();
    }
    if (TaskShoot.getIterations() == -1 && digitalRead(SHOOT_BUTTON))
    {
      TaskShoot.disable();
    }

  }

}
```

The callback is pretty complex because there are full-auto burst and semi-automatic weapons. So to make it short: the Task Callback sends an IR-packet and checks if the task should stay enabled to keep shooting or disables it self.

The Problem is when you keep pressing the shoot-button the ESP8266 will crash with the following exception: 

[... 123 lines cut ...]
0x40104b4c in lmacProcessTXStartData at ??:?
0x401064f1 in ets_timer_disarm at ??:?
0x40100634 in i2s_slc_isr at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_i2s.cpp:167
0x4022f928 in spiffs_page_data_check at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:28
0x40232134 in spiffs_object_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:2053
0x4022e529 in spiffs_hydro_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_hydrogen.cpp:417
0x4022ec42 in SPIFFS_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_hydrogen.cpp:430
0x4022c2e8 in spiffs_impl::SPIFFSFileImpl::read(unsigned char*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs_api.h:395 (discriminator 1)
0x40237510 in fs::File::read(unsigned char*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.cpp:73
0x4022646c in AudioFileSourceFS::read(void*, unsigned int) at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioFileSourceFS.cpp:49
0x40237260 in AudioGeneratorWAV::GetBufferedData(int, void*) at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioGeneratorWAV.cpp:67 (discriminator 4)
0x402266d0 in AudioGeneratorWAV::loop() at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioGeneratorWAV.cpp:98
0x4020427e in loopAudio() at src/main.cpp:480
0x40235f8a in std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:293
0x40100101 in std::function<void ()>::operator()() const at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:623
0x4020500a in Scheduler::execute() at .pio/libdeps/d1_mini_debug/TaskScheduler/src/TaskScheduler.h:1554
0x4020503f in loop at src/main.cpp:311
0x4022b062 in loop_wrapper() at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:258
0x40101775 in cont_wrapper at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81

```

The full source code of my project can be found <a href="https://github.com/CheatingBoy6890/Open-Lasertag-System-Tagger">here</a>.

Thank you for you're help.

arkhipenko, 2024-08-13:
I think the problem is the ISR code - it is typically advisable to do as little as possible in it:

```
void IRAM_ATTR onShoot()
...
    TaskShoot.restartDelayed(weapons[myWeaponIndex].getTimeBeforeShot());
...
} 
```

The code for `getTimeBeforeShot()` may not be in memory during ISR call, and an attempt to load it from flash fails. 
I would use values instead of method calls and maybe place those in IRAM as well.

CheatingBoy6890, 2024-08-13:
I tried replacing the 

```
TaskShoot.restartDelayed(...);
```

with just

```
TaskShoot.restart();
```
and then with a 2 millisecond delay (I know you shouldn't use delay in interrupts). But even then the exception states that exception stays the same:
```Exception (0):
epc1=0x40236240 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Illegal instruction
  epc1=0x40236240 in Task::isEnabled() at .pio/libdeps/d1_mini_debug/TaskScheduler/src/TaskScheduler.h:512

>>>stack>>>

ctx: cont
sp: 3ffff900 end: 3fffffc0 offset: 0150
3ffffa50:  40100a28 f0bb379c 77d67f28 40100add  
3ffffa60:  00000020 3ffffa90 3fff0d4c 40242fc8  
3ffffa70:  c0034034 3fff32f4 3ffffab0 402584f2  
3ffffa80:  401006a0 00000002 00000001 00000022  
3ffffa90:  3fffc200 401009ec 3fffc258 4000050c  
3ffffaa0:  4000437d 00000030 00000016 ffffffff  
3ffffab0:  60000200 00000008 ffffffff 80000000  
3ffffac0:  20000000 3fff2140 80000000 20344060  
3ffffad0:  80000000 00000002 00000734 3fff2144  
3ffffae0:  000000a0 00344060 00344000 00000030  
3ffffaf0:  401006a0 401005f8 3fffc270 4000050c  
3ffffb00:  400043e9 401005f8 3fffc270 4000050c  
3ffffb10:  400043a3 00000030 00000016 ffffffff  
3ffffb20:  60000200 00000006 1c261ac4 80000000  
3ffffb30:  20000000 3fff2400 80000000 203438e0  
3ffffb40:  00000000 00000004 000006a4 3fff2404  
3ffffb50:  00000020 003438e0 00343800 00000030  
3ffffb60:  40004b31 3fff1ebc 00000100 00338400  
3ffffb70:  40107070 00000030 0000001c ffffffff  
3ffffb80:  40100779 3fff1ebc 00000100 3fff1ebc  
3ffffb90:  00000000 400042db 00338400 40227575  
3ffffba0:  40004b31 3fff20e4 00000100 00344000  
3ffffbb0:  40107070 00000000 00000020 40100724  
3ffffbc0:  40100779 3fff20e4 00000100 3fff20e4  
3ffffbd0:  3ffe9e05 40105c1b 00344000 40227575  
3ffffbe0:  4010350f 3ffed868 0000000a 40227611  
3ffffbf0:  00007fff 0097443e 3ffee124 401036ec  
3ffffc00:  00000000 400042db 00000000 3fff0000  
3ffffc10:  40004b31 3fff1fd0 00000100 00348100  
3ffffc20:  40107070 7fffffff 00000000 00000001  
3ffffc30:  40100779 3fff1fd0 00000100 3fff1fd0  
3ffffc40:  3ffea6a4 00000000 00348100 40227575  
3ffffc50:  401006a0 2c9f0300 4000050c 40227611  
3ffffc60:  401038c8 401005f8 3fffc270 4000050c  
3ffffc70:  401006a0 00000030 0000001c ffffffff  
3ffffc80:  402358c4 401005f8 3fffc270 4000050c  
3ffffc90:  00000000 00000000 0000001f 40100724  
3ffffca0:  4022fdf2 00002000 3fffc228 401064b5  
3ffffcb0:  4000050c 00002000 00000429 00000100  
3ffffcc0:  00000100 00000030 00000017 ffffffff  
3ffffcd0:  4023266c 00000002 00010000 00000100  
3ffffce0:  401006a0 3fff1ebc 00338400 4022e29e  
3ffffcf0:  00000000 00000000 0000001f 40100724  
3ffffd00:  00000100 00000030 3fffc228 401064b5  
3ffffd10:  00000005 3ffffd50 3fff22b0 00000100  
3ffffd20:  00000100 3fff20e4 00344000 4022e29e  
3ffffd30:  00344000 3fff1e94 3fff17b4 4022bf60  
3ffffd40:  3fff17b4 00038400 00000384 40232723  
[... 84 lines cut ...]
0x4022fdce in spiffs_obj_lu_find_entry_visitor at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:168
0x4022f684 in spiffs_obj_lu_find_id_and_span_v at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:509
0x401006a0 in i2s_slc_isr at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_i2s.cpp:194
0x402302a2 in spiffs_obj_lu_find_id_and_span at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:539 (discriminator 4)
0x40232024 in spiffs_object_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_nucleus.cpp:2014
0x401005f8 in i2s_slc_isr at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_i2s.cpp:167
0x4025cc45 in cont_continue at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:51
0x4022e505 in spiffs_hydro_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_hydrogen.cpp:417
0x4022ec1e in SPIFFS_read at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs/spiffs_hydrogen.cpp:430
0x4022c2c4 in spiffs_impl::SPIFFSFileImpl::read(unsigned char*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/spiffs_api.h:395 (discriminator 1)
0x402374ec in fs::File::read(unsigned char*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.cpp:73
0x40226448 in AudioFileSourceFS::read(void*, unsigned int) at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioFileSourceFS.cpp:49
0x40227bf4 in HardwareSerial::write(unsigned char const*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/HardwareSerial.h:191
0x4023723c in AudioGeneratorWAV::GetBufferedData(int, void*) at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioGeneratorWAV.cpp:67 (discriminator 4)
0x40227bf4 in HardwareSerial::write(unsigned char const*, unsigned int) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/HardwareSerial.h:191
0x402266ac in AudioGeneratorWAV::loop() at .pio/libdeps/d1_mini_debug/ESP8266Audio/src/AudioGeneratorWAV.cpp:98
0x4020427e in loopAudio() at src/main.cpp:470
[... cut ...]
0x40100101 in std::function<void ()>::operator()() const at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:623
0x4020500a in Scheduler::execute() at .pio/libdeps/d1_mini_debug/TaskScheduler/src/TaskScheduler.h:1554
0x40205049 in loop at src/main.cpp:303
0x4022b03e in loop_wrapper() at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:258
0x40101739 in cont_wrapper at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81
``` 
So I think it's a problem with the TaskShoot.isEnabled method.

arkhipenko, 2024-08-13:
Possible. Can you try and get rid of it? Just restart (for now).

CheatingBoy6890, 2024-08-13:
I did and I had the same problem, but when I make the Task::isEnabled function an IRAM_ATTR as well as the TASK::restart function and the Task::enable functions the crash is getting really rare. So it seems that I have to make all the functions or methods IRAM_ATTR.

arkhipenko, 2024-08-13:
I thought about that - I will have to make it specific to ESP chips - this is not a problem on other architectures where the code is in memory all the time.  For now - go ahead and modify the library until I come up with a perm solution. 
I hit the same issue on esp32 for one of my projects - so this needs to be fixed.

CheatingBoy6890, 2024-08-13:
Ok thank you

CheatingBoy6890, 2024-08-13:
BTW it seems like the _task_millis() function needs to be in IRAM as well.

arkhipenko, 2024-08-13:
That's just crazy - but ok!
Seems line esp8266 is a lot more sensitive - esp32 does not have those issues.

arkhipenko, 2024-08-13:
just verified - esp32 `millis()` implementation is already `IRAM_ATTR`

CheatingBoy6890, 2024-08-14:
So I added the IRAM_ATTR to the isEnabled and other TaskScheduler functions. Sadly I noticed another problem: I'm using painlessmesh and as soon as I connect to another device in the mesh the ESP crashes this time with this exception: 

```
Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: sys
sp: 3ffffdd0 end: 3fffffb0 offset: 0010
3ffffde0:  00000000 00000000 3ffef8d0 4022b079  
3ffffdf0:  00000000 00000000 3ffef8d0 40205093  
3ffffe00:  00000000 5e76c8b4 0000864c 3ffef630  
3ffffe10:  3ffef5d4 00008657 3ffef8d0 40208749  
3ffffe20:  3ffef5d4 3fff0714 3fff0714 4022930c  
3ffffe30:  3fff3614 3fff0714 3ffef8d0 40212dbf  
3ffffe40:  00000000 00000004 feefeffe feefeffe  
3ffffe50:  00000000 00000000 00000000 00000000  
3ffffe60:  3fffff88 feefeffe 3fffff88 feefeffe  
3ffffe70:  3fff3544 feefeffe 40207b68 4020c714  
3ffffe80:  3fff3544 feefeffe 40207b08 4020c8ac  
3ffffe90:  00000000 73656d72 00000000 00000000  
3ffffea0:  64726f77 00000000 3fff2a0c feefeffe  
3ffffeb0:  40207b68 4020c714 04320085 ffffffff  
3ffffec0:  00000000 ed6ede05 3fff2a0c feefeffe  
3ffffed0:  40207b08 4020c8ac 00000437 feefeffe  
3ffffee0:  00000000 402360f8 3fff2a0c feefeffe  
3ffffef0:  40207c88 402075d8 00000437 feefeffe  
3fffff00:  00000000 402360f8 3fff2a0c feefeffe  
3fffff10:  402055dc 40204f24 00000451 feefeffe  
3fffff20:  00000000 40236130 40203090 feefeffe  
3fffff30:  40236178 40236164 4021300c feefeffe  
3fffff40:  402360d8 402360c4 40202d24 feefeffe  
3fffff50:  40236144 40236130 402164c0 feefeffe  
3fffff60:  40236110 402360f8 00000000 64726f77  
3fffff70:  00000000 00000000 73656d72 00000000  
3fffff80:  40203090 feefeffe 00000008 feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff08e0  
3fffffa0:  3fffdad0 00000000 3fffdab0 4022b163  
<<<stack<<<

0x4022b079 in yield at ??:?
0x40205093 in Scheduler::execute() at .pio/libdeps/d1_mini_debug/TaskScheduler/src/TaskScheduler.h:1480
0x40208749 in painlessmesh::Mesh<painlessmesh::Connection>::update() at .pio/libdeps/d1_mini_debug/painlessMesh/src/painlessmesh/mesh.hpp:162
 (inlined by) Connect_Vest() at src/main.cpp:816
0x4022930c in Print::println(__FlashStringHelper const*) at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Print.cpp:186
0x40212dbf in setup at src/main.cpp:255
0x40207b68 in std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#1}> const&, std::_Manager_operation) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:268
0x4020c714 in std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#1}>::_M_invoke(std::_Any_data const&, painlessmesh::protocol::Variant&&, std::shared_ptr<painlessmesh::Connection>&&, unsigned int&&) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40207b08 in std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#2}>::_M_manager(std::_Any_data&, std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#2}> const&, std::_Manager_operation) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:268
0x4020c8ac in std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#2}>::_M_invoke(std::_Any_data const&, painlessmesh::protocol::Variant&&, std::shared_ptr<painlessmesh::Connection>&&, unsigned int&&) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40207b68 in std::_Function_handler<void (painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int), painlessmesh::Mesh<painlessmesh::Connection>::onReceive(std::function<void (unsigned int, String&)>)::{lambda(painlessmesh::protocol::Variant, std::shared_ptr<painlessmesh::Connection>, unsigned int)#1}>::_M_manager(std::_Any_data&, std: [... cut ...]
0x402360f8 in std::_Function_handler<void (unsigned int, String&), void (*)(unsigned int, String&)>::_M_invoke(std::_Any_data const&, unsigned int&&, String&) at /home/silas/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40203090 in nodeTimeAdjustedCallback(int) at src/main.cpp:608
0x4022b163 in loop_wrapper() at /home/silas/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:255
```

What I think is happening (but could also be anything else) is that when the ESP connects to the mesh there are new Tasks created and because the methods from all these are in IRAM it overflows and the ESP crashes.

TD-er, 2024-08-14:
Or `delay()` is called from an ISR function.

arkhipenko, 2024-08-14:
I recently prefer to not do much in the ISRs at all - just set some volatile variable and restart an ISR servicing task that runs immediately - that way you are out of the interrupt context.

chrisalbertson, 2024-08-14:
> On Aug 14, 2024, at 2:05 PM, Anatoli Arkhipenko ***@***.***> wrote:
> 
> 
> I recently prefer to not do much in the ISRs at all - just set some volatile variable and restart an ISR servicing task that runs immediately - that way you are out of the interrupt context.
> 

This has been the best practice for as long as I can remember, going back to the mid-1970s.

That said there are some cases where people have reenable interrupts from within the ISR but that is just asking for trouble that is VERY hard to debug.

CheatingBoy6890, 2024-08-15:
So should I create another Task that is started instantly from the interrupt like
```cpp
//interrupt handler
void IRAM_ATTR onShoot(){
TaskHandleInterrupt.restart();
}
```
or just set a volatile variable and check if it changed in a Task?

CheatingBoy6890, 2024-08-15:
I think I fixed it now, the reason for the crash was that painlessmesh doesn't work in debug builds aka build_mode = debug.

arkhipenko, 2024-08-15:
> So should I create another Task that is started instantly from the interrupt like
Yes, you can do that, OR, if you want to check for stray events from time to time you can create a task that runs every N seconds (for instance) and, if running on a schedule checks for stray events (you need to figure out how). 
For the true interrupts:

```
//interrupt handler
volatile bool intOccurred = false; 
void IRAM_ATTR onShoot(){
intOccurred = true; 
TaskHandleInterrupt.forceNextIteration();
}
```

closed 2024-08-15 by arkhipenko