arkhipenko/TaskScheduler #191: Using esp_light_sleep_start() can break WiFi, should be configurable to use delay() instead.

ranma, 2024-09-29:
With the default of `esp_light_sleep_start()` I see WiFi connection problems.

Using delay() can still noticeably reduce the SoC temperature.  In mylimited testing, the temperature reported by `CpuTemperature.read()` reduced from ~58 °C to ~52 °C after enabling `-D_TASK_SLEEP_ON_IDLE_RUN=1` and hacking it to use `delay()` instead of `esp_light_sleep_start()`.

arkhipenko, 2024-09-29:
Hi,
Which version of TaskSxheduler are you running. 
I believe I disabled light sleep in the latest version since yes, it causes WiFi issues. 
Simple `delay(1)` (or `vTaskDelay(1)` ) works better.

ranma, 2024-09-29:
> Which version of TaskSxheduler are you running.

The master branch: https://github.com/arkhipenko/TaskScheduler/blob/master/src/TaskSchedulerSleepMethods.h#L61

ranma, 2024-10-06:
> Simple delay(1) (or vTaskDelay(1) ) works better.

One problem with a simple delay is that when using StatusRequest to wake from an interrupt, the delay is not interrupted.

So for ESP32 even better is to use `xTaskNotifyWait()` to sleep and `xTaskNotify()`/`xTaskNotifyFromISR()` to interrupt the sleep.

arkhipenko, 2025-09-21:
This would be very difficult to implement. I would try FreeRTOS tickless to do something like this in combination with the TASK_TICKLESS (example to follow)
Please submit a PR if you figure out how to make it work reliably