arkhipenko/TaskScheduler #193: crash esp8266

sanluca, 2024-12-31:
Hi, 
I'm experiencing crashes on my Wemos D1 mini when running code that uses the TaskScheduler library. I've checked the error logs, and I'm getting the following error messages related to TaskScheduler. I'd appreciate any help in troubleshooting this issue. 
Thank you.

```
15:59:55.697 -> COMMUNICATION: routePackage(): Recvd from 2575858913: {"nodeId":2575858913,"subs":[{"nodeId":756469405},{"nodeId":847376797,"root":true,"subs":[{"nodeId":864584653,"subs":[{"nodeId":2575841157,"subs":[{"nodeId":257SYNC: handleNodeSync(): with 2575858913
15:59:55.739 -> 
15:59:55.739 -> User exception (panic/abort/assert)
15:59:55.739 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:59:55.739 -> 
15:59:55.739 -> Panic core_esp8266_main.cpp:191 __yield
15:59:55.739 -> 
15:59:55.739 -> >>>stack>>>
15:59:55.739 -> 
15:59:55.739 -> ctx: sys
15:59:55.768 -> sp: 3fffff70 end: 3fffffb0 offset: 0010
15:59:55.768 -> 3fffff80:  01db01df 00000000 402174ce 402174f1  
15:59:55.768 -> 3fffff90:  3ffeff00 00000000 00000000 40210b9f  
15:59:55.768 -> 3fffffa0:  3fffdad0 00000000 3fff0430 3fff045c  
15:59:55.768 -> <<<stack<<<
15:59:55.768 -> 
15:59:55.768 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:59:55.768 -> 
15:59:55.768 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
15:59:55.808 -> 
15:59:55.808 -> load 0x4010f000, len 3424, room 16 
15:59:55.808 -> tail 0
15:59:55.808 -> chksum 0x2e
15:59:55.808 -> load 0x3fff20b8, len 40, room 8 

Decoding stack results
0x402174ce: __yield() at C:\Users\sanlu\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266\core_esp8266_features.h:64
0x40210b9f: Scheduler::execute() at c:\Users\sanlu\Documents\Arduino\libraries\TaskScheduler\src\TaskScheduler.h:1577

Paste exception to decode...
```

```
//************************************************************
//"caldaia" wemos d1 mini pro
//
// rele D1 --> 5
// 
// painlessmesh 1.5.4 arduinojson 7.3.0
//************************************************************
#include "namedMesh.h"
const byte relay = D1;
bool relayState = LOW;
const unsigned long RETRY_ROOT_INTERVAL = 1200000; // 20 minuti
bool retryTaskEnabled = false; // Flag per il task di retry

//#include "painlessMesh.h"

#define   MESH_PREFIX     "whateverYouLike"
#define   MESH_PASSWORD   "somethingSneaky"
#define   MESH_PORT       5555

Scheduler userScheduler; // to control your personal task
//painlessMesh  mesh;
namedMesh  mesh;
String msg,rel;
String nodeName = "caldaia"; // Name needs to be unique
String to = "bridgemqtt";
uint32_t root_id=0;

#define ROLE    "caldaia"
#define VERSION "Caldaia v3.0.0"
#define MESSAGE "caldaia "

// User stub
void sendMessage() ; // Prototype so PlatformIO doesn't complain
void retryRoot();

Task taskSendMessage( TASK_SECOND * 1 , TASK_FOREVER, &sendMessage );
Task retryRootTask(RETRY_ROOT_INTERVAL, TASK_ONCE, &retryRoot);

void sendMessage() {
    msg = "output/";
    msg += relayState;
    mesh.sendSingle(to, msg);
    update_status();
    taskSendMessage.setInterval( random( TASK_SECOND * 50, TASK_SECOND * 70 ));
}

// Needed for painless library
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("startHere: Received from %u msg=%s\n", from, msg.c_str());
  rel=msg;
  root_id=from;
 if (strcmp(rel.c_str(),"1") == 0){
      digitalWrite(relay, HIGH);   
      relayState = HIGH;
      msg = "output/1";
      mesh.sendSingle(to, msg);
      
   }
   else if (strcmp(rel.c_str(),"0") == 0) {
    digitalWrite(relay, LOW);   
    relayState = LOW;
    msg = "output/0";
    mesh.sendSingle(to, msg);
}}

void newConnectionCallback(uint32_t nodeId) {
    Serial.printf("--> startHere: New Connection, nodeId = %u\n", nodeId);
}

void changedConnectionCallback() {
  Serial.println("changedConnectionCallback");
  if (!mesh.isConnected(root_id)) {
    Serial.println("Connessione al root persa! Pianifico il retry.");
    if (!retryTaskEnabled) { // Controlla il flag
      retryRootTask.enable();
      retryTaskEnabled = true; // Imposta il flag
    }
  } else {
    Serial.println("Connessione al root ripristinata.");
    retryRootTask.disable();
    retryTaskEnabled = false; // Resetta il flag
}}

void nodeTimeAdjustedCallback(int32_t offset) {
    Serial.printf("Adjusted time %u. Offset = %d\n", mesh.getNodeTime(),offset);
}

void update_status() {
  long uptime = millis() / 60000L;
  msg = "uptime/";
  msg += uptime;
  mesh.sendSingle(to, msg);
  msg = "nodeid/";
  msg += mesh.getNodeId();
  mesh.sendSingle(to, msg);
  msg = "freememory/";
  msg += String(ESP.getFreeHeap());
  mesh.sendSingle(to, msg);
  msg = "version/";
  msg += VERSION;
  mesh.sendSingle(to, msg);
  msg = "root/";
  msg += root_id;
  mesh.sendSingle(to, msg);
  msg = "ip/";
  msg += WiFi.localIP().toString();
  mesh.sendSingle(to, msg);
  msg = "wifisignal/";
  msg += String(WiFi.RSSI());
  mesh.sendSingle(to, msg);
}

void retryRoot() {
  if (!mesh.isConnected(root_id)) {
    Serial.println("Ancora offline dopo il retry. Riavvio...");
    ESP.restart();
  } else {
    Serial.println("Root tornato online.");
    retryTaskEnabled = false; // Resetta il flag anche qui, per sicurezza
}}

void setup() {
  Serial.begin(115200);
  pinMode(relay, OUTPUT);

  mesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on
  //mesh.setDebugMsgTypes( ERROR | STARTUP );  // set before init() so that you can see startup messages
  mesh.init( MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT, WIFI_AP_STA, 11 );
  mesh.initOTAReceive(ROLE);
  mesh.setContainsRoot(true);
  mesh.setName(nodeName);
  mesh.onReceive(&receivedCallback);
  mesh.onNewConnection(&newConnectionCallback);
  mesh.onChangedConnections(&changedConnectionCallback);
  mesh.onNodeTimeAdjusted(&nodeTimeAdjustedCallback);

  userScheduler.addTask( taskSendMessage );
  taskSendMessage.enable();
}

void loop() {
  // it will run the user scheduler as well
  mesh.update(); 
}
```

CheatingBoy6890, 2025-02-02:
I recently had a similar error which was related to two problems: when I included painlessmesh and TaskScheduler in the same file it wouldn't work so I just included painlessmesh. Another problem was using debug build in platformio which also caused my D1 mini to crash.

arkhipenko, 2025-09-21:
Please open an issue with Painless mesh author.