arkhipenko/TaskScheduler #196: Time elapsed since the last execution of a task that has already finished its interactions.

georgevbsantiago, 2025-01-27:
Hello, Thank you for this awesome library.

I'm trying to understand if there is a way to get the time elapsed since the last execution of a task that has already finished its interactions.

It would be something similar/equal to the `elapsed()` method from the library [Ticker](https://github.com/sstaub/Ticker)

> Ticker Elapsed Time
> uint32_t Ticker::elapsed()
> Returns the time passed since the last tick in ms or us depending on mode.

In my use case, I'm checking a set of conditions and one of them is the time elapsed since the last execution of a task that has already finished its interactions.

The way I found to solve my problem:

```
unsigned long millis_check_OFF_to_on;

void uplink_imediato_function();
Task uplink_imediato_task(50, 1, &uplink_imediato_function, &loop_Tasks_Scheduler, false);

if(status_current_previous == false &&
   ((millis() - millis_check_OFF_to_on) >= ( 1 * TASK_MINUTE))) {

   uplink_imediato_task.restart();

   millis_check_OFF_to_on = millis();

}
```

If there was an approach similar to `elapsed()` from the Ticker library, then I could use this way:

```
void uplink_imediato_function();
Task uplink_imediato_task(50, 1, &uplink_imediato_function, &loop_Tasks_Scheduler, false);

if(status_current_previous == false &&
  uplink_imediato_task.elapsed() >= ( 1 * TASK_MINUTE))) {

   uplink_imediato_task.restart();

}
```

Is there something similar to Ticker's `elapsed()` method in the TaskScheduler library?

georgevbsantiago, 2025-01-28:
In the case mentioned above, the objective is to obtain the time elapsed since the last execution of a task that has already finished its interactions.

However, from the context above it would be possible to extract two new implementations:

- Time elapsed since the last execution of a task that has already finished its interactions;
- Time elapsed since the last iteration of a task that is still active and running;

Thank you for considering this feature.
It will help in tasks that involve performing uplink, as illustrated in the code above.

arkhipenko, 2025-01-28:
I think it is relatively easy for me to capture the exact moment ( `millis()` ) when the task's callback was invoked (entered) and finished (returned to scheduler). Same for `onEnable()` and `onDisable()`
I believe all other stats could be calculated from there. 
Do you agree?

georgevbsantiago, 2025-01-29:
Perhaps the first case (which was the one in my example) is the easiest to implement:
- Time elapsed since the last execution of a task that has already finished its interactions;

The second case may have many exceptions and complexities.
- Time elapsed since the last iteration of a task that is still active and running;

I believe that the first case would already be an excellent feature to be added.

misc-protolabs, 2025-01-29:
for better resolution - micros() is a better choice.  same rollover issues, but still better timing resolution...

arkhipenko, 2025-01-29:
> You can assign -1 to tasks that have never been executed or activated.

Cannot use negative numbers - `millis()` is `uint32_t`

arkhipenko, 2025-09-21:
Actually, this is very easy to implement in your code without updates to the library:
You can use `TASK_LTS` - local task storage pointer to store the `millis()` value when the task is disabled, using the `onDisable()` method, and then compare to that value if the tast is not enabled anymore. 

```
Scheduler ts;
Task t(100,2,&callback,&ts,true, null, &callbackOD); 

void callbackOD() {
   uint32_t* t = (uint32_t*) ts.currentLts();
   *t = millis();
}
```

then in your main code:
```
if ( !uplink_imediato_task.isEnabled() && (millis() - (uint32_t) uplink_imediato_task.getLtsPointer()) > TASK MINUTE) ...
```

closed 2025-09-21 by arkhipenko