arkhipenko/TaskScheduler #200: warning: using value of assignment with 'volatile'

hoffmakl1961, 2025-04-01:
Hello everyone,
I get the following warning when compiling: 

In file included from C:\Users\...\Documents\Arduino\Heizung\Thermostat\Thermostat_Wohnen_V12\Thermostat_Wohnen_V12.ino:22:
c:\Users\...\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Task::set(long unsigned int, long int, TaskCallback, TaskOnEnable, TaskOnDisable)':
c:\Users\...\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:626:34: warning: using value of assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  626 |     iSetIterations = iIterations = aIterations;

c:\Users\...\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h: In member function 'void Task::setIterations(long int)':
c:\Users\...\Documents\Arduino\libraries\TaskScheduler\src/TaskScheduler.h:642:34: 
warning: using value of assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  642 |     iSetIterations = iIterations = aIterations;
      |                      ~~~~~~~~~~~~^~~~~~~~~~~~~

Mein Environment: 
Ardunino IDE 2.3.4 / ESP32 Dev Module, TaskScheduler 3.8.5 ([arkhipenko]
let me know, if you need more informations

hoffmakl1961, 2025-04-04:
The compiler warns that an assignment to a volatile-qualified variable is no longer recommended. This means that the expression:
_iSetIterations = iIterations = aIterations;_
is problematic if iIterations has been declared as volatile.

Why does the error occur?
- volatile is used to inform the compiler that a variable can change unexpectedly (e.g. due to interrupts or external hardware).
- In newer C++ standards, direct chained assignment (=) with volatile variables is considered obsolete as it can lead to unpredictable behaviour.

Solution:
Instead of writing the assignment in one line, the values should be assigned one after the other:
_iIterations = aIterations; // Set the value first
iSetIterations = iIterations; // Then assign the set value_

TaskScheduler.h
line 626 and 642

arkhipenko, 2025-09-21:
Fixed in v4.0.0

closed 2025-09-21 by arkhipenko